intel/compute-runtime #762: Cannot save cache to other directory on Windows using NEO_CACHE_DIR

CarlGao4, 2024-09-06:
I set `NEO_CACHE_DIR` to another folder on Windows, and monitored the program with process monitor. I only saw that the program read the folder twice and no cache was written or read. However, if I remove this environment variable, cache was saved to `%localappdata%\NEO\neo_compiler_cache`. Driver version: `32.0.15.6081`
I tested using intel extension for pytorch.

---

No environment variable:
[image]

Previous files removed before environment variable set:
[image]
I've make sure that the path could be written in the latter test

JablonskiMateusz, 2024-09-09:
Hi @CarlGao4  looks like an issue in reading attributes on our side. Fix in progress

closed 2024-09-09 by VPG-SWE-Github

CarlGao4, 2024-09-10:
I'd like to ask when can I see this update in the graphic driver?

JablonskiMateusz, 2024-09-10:
this will be included in 24.39 release

CarlGao4, 2024-09-10:
Sorry for disturbing, but how can I know which version am I using on Windows?

commit e30b8c03 (2024-09-09): fix: windows cl_cache fixes and improvements

Fixed the function that checks if a directory exists.
Added debug message in case config file creation fails.
Simplified setting pattern for searching all files in a directory.
Adding file search loop skip when found file is a directory.

Fixes: #762