intel/compute-runtime #763: ODR violation issue in  24.26.30049.6

bl4ckb0ne, 2024-09-10:
Hi,

I'm currently packaging the project for Alpine Linux (see [1]) and I'm encountering ODR violation issue at the end of the build. I picked this version because of issue with BMG and LUNARLAKE in the latest version.

Here's the logs:

```
ninja: job failed: : && /usr/bin/ccache /usr/lib/ccache/bin/g++ -fPIC -Os -fstack-clash-protection -Wformat -Werror=format-security -D_GLIBCXX_ASSERTIONS=1 -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -D_LIBCPP_ENABLE_HARDENED_MODE=1 -fno-plt -flto=auto -ffat-lto-objects -mindirect-branch=thunk -mfunction-return=thunk -mindirect-branch-register -ftemplate-depth=1024 -mclflushopt -mwaitpkg -Wall -Wempty-body -Wignored-qualifiers -Wtype-limits -Wuninitialized -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Winvalid-pch -Wno-unknown-pragmas -Wno-unused-local-typedefs -Wno-unused-but-set-variable -Wclobbered -Wimplicit-fallthrough=4 -Wno-noexcept-type -Werror -Werror=vla  -Wl,--as-needed,-O1,--sort-common -Wl,-z,pack-relative-relocs   -Wl,--version-script=/home/simon/aports/testing/intel-opencl-icd/src/compute-runtime-24.26.30049.6/build/igdrcl_lib_release/ocl.exports -Wl,-Bsymbolic -shared -Wl,-soname,libigdrcl.so -o bin/libigdrcl.so @CMakeFiles/igdrcl_dll.rsp && :
/usr/include/igdgmm/inc/umKmInc/../common/sku_wa.h:187:16: error: type 'struct _WA_TABLE' violates the C++ One Definition Rule [-Werror=odr]
  187 | typedef struct _WA_TABLE
      |                ^
/usr/include/igdgmm/inc/umKmInc/../common/sku_wa.h:187:16: note: a different type is defined in another translation unit
  187 | typedef struct _WA_TABLE
      |                ^
/usr/include/igdgmm/inc/umKmInc/../common/sku_wa.h:190:24: note: the first difference of corresponding definitions is field '<anonymous>'
  190 |         unsigned int : 0;
      |                        ^
/usr/include/igdgmm/inc/umKmInc/../common/sku_wa.h:190:24: note: a field of same name but different type is defined in another translation unit
  190 |         unsigned int : 0;
      |                        ^
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:79:29: error: type 'struct Context' violates the C++ One Definition Rule [-Werror=odr]
   79 |     class NON_PAGED_SECTION Context : public GmmMemAllocator
      |                             ^
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:79:29: note: a different type is defined in another translation unit
   79 |     class NON_PAGED_SECTION Context : public GmmMemAllocator
      |                             ^
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:93:42: note: the first difference of corresponding definitions is field 'WaTable'
   93 |         WA_TABLE                         WaTable;
      |                                          ^
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:93:42: note: a field of same name but different type is defined in another translation unit
   93 |         WA_TABLE                         WaTable;
      |                                          ^
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:79:29: note: type 'struct WA_TABLE' itself violates the C++ One Definition Rule
   79 |     class NON_PAGED_SECTION Context : public GmmMemAllocator
      |                             ^
/home/simon/aports/testing/intel-opencl-icd/src/compute-runtime-24.26.30049.6/shared/source/os_interface/windows/sharedata_wrapper.h:102: error: type 'struct WA_TABLE_KMD' violates the C++ One Definition Rule [-Werror=odr]
  102 | struct WA_TABLE_KMD : WA_TABLE_GMM { // NOLINT(readability-identifier-naming)
/home/simon/aports/testing/intel-opencl-icd/src/compute-runtime-24.26.30049.6/shared/source/os_interface/windows/sharedata_wrapper.h:102: note: a type with different bases is defined in another translation unit
  102 | struct WA_TABLE_KMD : WA_TABLE_GMM { // NOLINT(readability-identifier-naming)
/usr/include/igdgmm/inc/umKmInc/sharedata.h:166: error: type 'struct _ADAPTER_INFO' violates the C++ One Definition Rule [-Werror=odr]
  166 | typedef struct _ADAPTER_INFO
/usr/include/igdgmm/inc/umKmInc/sharedata.h:166: note: a different type is defined in another translation unit
  166 | typedef struct _ADAPTER_INFO
/usr/include/igdgmm/inc/umKmInc/sharedata.h:172: note: the first difference of corresponding definitions is field 'WaTable'
  172 |     WA_TABLE                    WaTable;                // WA table
/usr/include/igdgmm/inc/umKmInc/sharedata.h:172: note: a field of same name but different type is defined in another translation unit
  172 |     WA_TABLE                    WaTable;                // WA table
/usr/include/igdgmm/inc/umKmInc/sharedata.h:166: note: type 'struct WA_TABLE' itself violates the C++ One Definition Rule
  166 | typedef struct _ADAPTER_INFO
/home/simon/aports/testing/intel-opencl-icd/src/compute-runtime-24.26.30049.6/shared/source/os_interface/windows/sharedata_wrapper.h:179: error: type 'struct ADAPTER_INFO_KMD' violates the C++ One Definition Rule [-Werror=odr]
  179 | struct ADAPTER_INFO_KMD : ADAPTER_INFO_GMM { // NOLINT(readability-identifier-naming)
/home/simon/aports/testing/intel-opencl-icd/src/compute-runtime-24.26.30049.6/shared/source/os_interface/windows/sharedata_wrapper.h:179: note: a type with different bases is defined in another translation unit
  179 | struct ADAPTER_INFO_KMD : ADAPTER_INFO_GMM { // NOLINT(readability-identifier-naming)
lto1: all warnings being treated as errors
lto-wrapper: fatal error: /usr/bin/g++ returned 1 exit status
compilation terminated.
/usr/lib/gcc/x86_64-alpine-linux-musl/14.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
ninja: subcommand failed
```

I noticed it's coming from the `windows` part but I'm building on Linux, is this caused by an error somewhere in CMake? The error also occurs with `24.22.29735.20`, intel-gmmlib version is `22.5.1`.

Thanks

[1]: https://gitlab.alpinelinux.org/alpine/aports/-/merge_requests/71645

JablonskiMateusz, 2024-09-11:
Hi @bl4ckb0ne 
Windows part here may be related to WSL support. You may disable that by setting cmake flag `DISABLE_WDDM_LINUX=1`

JablonskiMateusz, 2024-09-11:
could you narrow-down which cpp files causes the definition mismatch ?

bl4ckb0ne, 2024-09-11:
@JablonskiMateusz still failing with ` -DDISABLE_WDDM_LINUX=ON`, same error.

```
ninja: job failed: : && /usr/bin/ccache /usr/lib/ccache/bin/g++ -fPIC -Os -fstack-clash-protection -Wformat -Werror=format-security -D_GLIBCXX_ASSERTIONS=1 -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -D_LIBCPP_ENABLE_HARDENED_MODE=1 -fno-plt -flto=auto -ffat-lto-objects -mindirect-branch=thunk -mfunction-return=thunk -mindirect-branch-register -ftemplate-depth=1024 -mclflushopt -mwaitpkg -Wall -Wempty-body -Wignored-qualifiers -Wtype-limits -Wuninitialized -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Winvalid-pch -Wno-unknown-pragmas -Wno-unused-local-typedefs -Wno-unused-but-set-variable -Wclobbered -Wimplicit-fallthrough=4 -Wno-noexcept-type -Werror -Werror=vla  -Wl,--as-needed,-O1,--sort-common -Wl,-z,pack-relative-relocs   -Wl,--version-script=/home/simon/aports/testing/intel-opencl-icd/src/compute-runtime-24.26.30049.6/build/igdrcl_lib_release/ocl.exports -Wl,-Bsymbolic -shared -Wl,-soname,libigdrcl.so -o bin/libigdrcl.so @CMakeFiles/igdrcl_dll.rsp && :
/usr/include/igdgmm/GmmLib/inc/../../inc/umKmInc/../common/sku_wa.h:187:16: error: type 'struct _WA_TABLE' violates the C++ One Definition Rule [-Werror=odr]
  187 | typedef struct _WA_TABLE
      |                ^
/usr/include/igdgmm/GmmLib/inc/../../inc/umKmInc/../common/sku_wa.h:187:16: note: a different type is defined in another translation unit
  187 | typedef struct _WA_TABLE
      |                ^
/usr/include/igdgmm/GmmLib/inc/../../inc/umKmInc/../common/sku_wa.h:190:24: note: the first difference of corresponding definitions is field '<anonymous>'
  190 |         unsigned int : 0;
      |                        ^
/usr/include/igdgmm/GmmLib/inc/../../inc/umKmInc/../common/sku_wa.h:190:24: note: a field of same name but different type is defined in another translation unit
  190 |         unsigned int : 0;
      |                        ^
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:79:29: error: type 'struct Context' violates the C++ One Definition Rule [-Werror=odr]
   79 |     class NON_PAGED_SECTION Context : public GmmMemAllocator
      |                             ^
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:79:29: note: a different type is defined in another translation unit
   79 |     class NON_PAGED_SECTION Context : public GmmMemAllocator
      |                             ^
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:93:42: note: the first difference of corresponding definitions is field 'WaTable'
   93 |         WA_TABLE                         WaTable;
      |                                          ^
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:93:42: note: a field of same name but different type is defined in another translation unit
   93 |         WA_TABLE                         WaTable;
      |                                          ^
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:79:29: note: type 'struct WA_TABLE' itself violates the C++ One Definition Rule
   79 |     class NON_PAGED_SECTION Context : public GmmMemAllocator
      |                             ^
lto1: all warnings being treated as errors
lto-wrapper: fatal error: /usr/bin/g++ returned 1 exit status
compilation terminated.
/usr/lib/gcc/x86_64-alpine-linux-musl/14.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
ninja: subcommand failed
```

Seems to happen during linking, could it be an issue with LTO?

bl4ckb0ne, 2024-09-11:
I tried the option `-DNEO_DISABLE_LD_GOLD:BOOL=ON` like the arch package does and got more log

```
ninja: job failed: : && /usr/bin/ccache /usr/lib/ccache/bin/g++ -fPIC -Os -fstack-clash-protection -Wformat -Werror=format-security -D_GLIBCXX_ASSERTIONS=1 -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -D_LIBCPP_ENABLE_HARDENED_MODE=1 -fno-plt -flto=auto -ffat-lto-objects -mindirect-branch=thunk -mfunction-return=thunk -mindirect-branch-register -ftemplate-depth=1024 -mclflushopt -mwaitpkg -Wall -Wempty-body -Wignored-qualifiers -Wtype-limits -Wuninitialized -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Winvalid-pch -Wno-unknown-pragmas -Wno-unused-local-typedefs -Wno-unused-but-set-variable -Wclobbered -Wimplicit-fallthrough=4 -Wno-noexcept-type -Werror -Werror=vla -fpermissive -fPIC  -Wl,--as-needed,-O1,--sort-common -Wl,-z,pack-relative-relocs   -Wl,--version-script=/home/simon/aports/testing/intel-opencl-icd/src/compute-runtime-24.26.30049.6/level_zero/core/source/dll/linux/ze.exports -shared -Wl,-soname,libze_intel_gpu.so.1 -o bin/libze_intel_gpu.so.1.3.30049 @CMakeFiles/ze_intel_gpu.rsp && :
lto1: fatal error: Cannot open shared/source/built_ins/CMakeFiles/builtins_binaries_bindful.dir/__/__/__/bin/built_ins/x64/gen11/bindful_copy_buffer_to_image3d_11_1_0.cpp.o
compilation terminated.
lto-wrapper: fatal error: /usr/bin/g++ returned 1 exit status
compilation terminated.
/usr/lib/gcc/x86_64-alpine-linux-musl/14.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
ninja: job failed: : && /usr/bin/ccache /usr/lib/ccache/bin/g++ -fPIC -Os -fstack-clash-protection -Wformat -Werror=format-security -D_GLIBCXX_ASSERTIONS=1 -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -D_LIBCPP_ENABLE_HARDENED_MODE=1 -fno-plt -flto=auto -ffat-lto-objects -mindirect-branch=thunk -mfunction-return=thunk -mindirect-branch-register -ftemplate-depth=1024 -mclflushopt -mwaitpkg -Wall -Wempty-body -Wignored-qualifiers -Wtype-limits -Wuninitialized -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Winvalid-pch -Wno-unknown-pragmas -Wno-unused-local-typedefs -Wno-unused-but-set-variable -Wclobbered -Wimplicit-fallthrough=4 -Wno-noexcept-type -Werror -Werror=vla  -Wl,--as-needed,-O1,--sort-common -Wl,-z,pack-relative-relocs   -Wl,--version-script=/home/simon/aports/testing/intel-opencl-icd/src/compute-runtime-24.26.30049.6/build/igdrcl_lib_release/ocl.exports -Wl,-Bsymbolic -shared -Wl,-soname,libigdrcl.so -o bin/libigdrcl.so @CMakeFiles/igdrcl_dll.rsp && :
/usr/include/igdgmm/GmmLib/inc/../../inc/umKmInc/../common/sku_wa.h:187:16: error: type 'struct _WA_TABLE' violates the C++ One Definition Rule [-Werror=odr]
  187 | typedef struct _WA_TABLE
      |                ^
/usr/include/igdgmm/GmmLib/inc/../../inc/umKmInc/../common/sku_wa.h:187:16: note: a different type is defined in another translation unit
  187 | typedef struct _WA_TABLE
      |                ^
/usr/include/igdgmm/GmmLib/inc/../../inc/umKmInc/../common/sku_wa.h:190:24: note: the first difference of corresponding definitions is field '<anonymous>'
  190 |         unsigned int : 0;
      |                        ^
/usr/include/igdgmm/GmmLib/inc/../../inc/umKmInc/../common/sku_wa.h:190:24: note: a field of same name but different type is defined in another translation unit
  190 |         unsigned int : 0;
      |                        ^
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:79:29: error: type 'struct Context' violates the C++ One Definition Rule [-Werror=odr]
   79 |     class NON_PAGED_SECTION Context : public GmmMemAllocator
      |                             ^
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:79:29: note: a different type is defined in another translation unit
   79 |     class NON_PAGED_SECTION Context : public GmmMemAllocator
      |                             ^
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:93:42: note: the first difference of corresponding definitions is field 'WaTable'
   93 |         WA_TABLE                         WaTable;
      |                                          ^
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:93:42: note: a field of same name but different type is defined in another translation unit
   93 |         WA_TABLE                        ninja: job failed: : && /usr/bin/ccache /usr/lib/ccache/bin/g++ -fPIC -Os -fstack-clash-protection -Wformat -Werror=format-security -D_GLIBCXX_ASSERTIONS=1 -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -D_LIBCPP_ENABLE_HARDENED_MODE=1 -fno-plt -flto=auto -ffat-lto-objects -mindirect-branch=thunk -mfunction-return=thunk -mindirect-branch-register -ftemplate-depth=1024 -mclflushopt -mwaitpkg -Wall -Wempty-body -Wignored-qualifiers -Wtype-limits -Wuninitialized -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Winvalid-pch -Wno-unknown-pragmas -Wno-unused-local-typedefs -Wno-unused-but-set-variable -Wclobbered -Wimplicit-fallthrough=4 -Wno-noexcept-type -Werror -Werror=vla -fpermissive -fPIC  -Wl,--as-needed,-O1,--sort-common -Wl,-z,pack-relative-relocs   -Wl,--version-script=/home/simon/aports/testing/intel-opencl-icd/src/compute-runtime-24.26.30049.6/level_zero/core/source/dll/linux/ze.exports -shared -Wl,-soname,libze_intel_gpu.so.1 -o bin/libze_intel_gpu.so.1.3.30049 @CMakeFiles/ze_intel_gpu.rsp && :
[... cut ...]
/usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h:79:29: note: type 'struct WA_TABLE' itself violates the C++ One Definition Rule
   79 |     class NON_PAGED_SECTION Context : public GmmMemAllocator
      |                             ^
lto1: all warnings being treated as errors
lto-wrapper: fatal error: /usr/bin/g++ returned 1 exit status
compilation terminated.
/usr/lib/gcc/x86_64-alpine-linux-musl/14.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
ninja: subcommands failed
```

bl4ckb0ne, 2024-09-11:
`-DNEO_DISABLE_LTO=ON ` worked!

JablonskiMateusz, 2024-09-16:
@bl4ckb0ne build with LTO is optional, can be disabled with the flag you mentioned. However I still wonder which cpp files introduced conflicting definitions. Based on the log you shared:
> /usr/include/igdgmm/GmmLib/inc/../../inc/umKmInc/../common/sku_wa.h:190:24: note: the first difference of corresponding definitions is field '<anonymous>'
  190 |         unsigned int : 0;
      |                        ^
/usr/include/igdgmm/GmmLib/inc/../../inc/umKmInc/../common/sku_wa.h:190:24: note: a field of same name but different type is defined in another translation unit
  190 |         unsigned int : 0;

it is difference on `unsigned int` definition

JablonskiMateusz, 2024-09-16:
could you try to patch sku_wa.h to remove `unsigned int : 0; ` lines?

bl4ckb0ne, 2024-09-16:
I had to remove both https://github.com/intel/gmmlib/blob/master/Source/inc/common/sku_wa.h#L197 and https://github.com/intel/gmmlib/blob/master/Source/inc/common/sku_wa.h#L206 to build with LTO. Should I report the issue there?

JablonskiMateusz, 2024-09-17:
@johnmach please take a look

johnmach, 2024-10-03:
@bl4ckb0ne
Thanks for raising this. 

Could you please try to build using this patch:
https://github.com/intel/gmmlib/pull/123
This PR removes the repetitive declaration, but maintains struct alignment.

bl4ckb0ne, 2024-10-21:
No luck

```
ninja: job failed: /usr/bin/g++ @shared/source/CMakeFiles/neo_shared.dir/os_interface/linux/drm_memory_manager_create_multi_host_allocation.cpp.o.rsp -MD -MT shared/source/CMakeFiles/neo_shared.dir/os_interface/linux/drm_memory_manager_create_multi_host_allocation.cpp.o -MF shared/source/CMakeFiles/neo_shared.dir/os_interface/linux/drm_memory_manager_create_multi_host_allocation.cpp.o.d -o shared/source/CMakeFiles/neo_shared.dir/os_interface/linux/drm_memory_manager_create_multi_host_allocation.cpp.o -c /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/shared/source/os_interface/linux/drm_memory_manager_create_multi_host_allocation.cpp
In file included from /usr/include/c++/14.2.0/vector:66,
                 from /usr/include/c++/14.2.0/functional:64,
                 from /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/shared/source/helpers/aligned_memory.h:15,
                 from /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/shared/source/os_interface/linux/drm_memory_manager_create_multi_host_allocation.cpp:9:
In member function 'std::vector<NEO::BufferObject*, std::allocator<NEO::BufferObject*> >::push_back(NEO::BufferObject* const&)',
    inlined from 'StackVec<NEO::BufferObject*, 4ul, unsigned char>::push_back(NEO::BufferObject* const&)' at /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/shared/source/utilities/stackvec.h:225:34,
    inlined from 'StackVec<NEO::BufferObject*, 4ul, unsigned char>::push_back(NEO::BufferObject* const&)' at /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/shared/source/utilities/stackvec.h:221:10,
    inlined from 'NEO::DrmAllocation::DrmAllocation(unsigned int, unsigned long, NEO::AllocationType, NEO::BufferObject*, void*, unsigned long, unsigned long, NEO::MemoryPool)' at /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/shared/source/os_interface/linux/drm_allocation.h:48:32:
/usr/include/c++/14.2.0/bits/stl_vector.h:1285:27: error: 'MEM[(struct vector *)this_10(D) + 1528B].D.185593._M_impl.D.184929._M_finish' may be used uninitialized [-Werror=maybe-uninitialized]
 1285 |         if (this->_M_impl._M_finish != this->_M_impl._M_end_of_storage)
      |             ~~~~~~~~~~~~~~^~~~~~~~~
/usr/include/c++/14.2.0/bits/stl_vector.h:1285:54: error: 'MEM[(struct vector *)this_10(D) + 1528B].D.185593._M_impl.D.184929._M_end_of_storage' may be used uninitialized [-Werror=maybe-uninitialized]
 1285 |         if (this->_M_impl._M_finish != this->_M_impl._M_end_of_storage)
      |                                        ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
ninja: job failed: /usr/bin/g++ @igdrcl_lib_release/CMakeFiles/igdrcl_lib_release.dir/built_ins/vme_builtin.cpp.o.rsp -MD -MT igdrcl_lib_release/CMakeFiles/igdrcl_lib_release.dir/built_ins/vme_builtin.cpp.o -MF igdrcl_lib_release/CMakeFiles/igdrcl_lib_release.dir/built_ins/vme_builtin.cpp.o.d -o igdrcl_lib_release/CMakeFiles/igdrcl_lib_release.dir/built_ins/vme_builtin.cpp.o -c /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/opencl/source/built_ins/vme_builtin.cpp
In file included from /usr/include/c++/14.2.0/vector:72,
                 from /usr/include/c++/14.2.0/functional:64,
                 from /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/shared/source/compiler_interface/compiler_options.h:13,
                 from /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/shared/source/built_ins/built_ins.h:11,
                 from /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/opencl/source/built_ins/vme_builtin.cpp:10:
In member function 'std::vector<NEO::ClDevice*, std::allocator<NEO::ClDevice*> >::emplace_back<NEO::ClDevice*>(NEO::ClDevice*&&)NEO::ClDevice*&',
    inlined from 'std::vector<NEO::ClDevice*, std::allocator<NEO::ClDevice*> >::push_back(NEO::ClDevice*&&)' at /usr/include/c++/14.2.0/bits/stl_vector.h:1301:21,
    inlined from 'StackVec<NEO::ClDevice*, 1ul, unsigned char>::push_back(NEO::ClDevice*&&)' at /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/shared/source/utilities/stackvec.h:237:34,
    inlined from 'StackVec<NEO::ClDevice*, 1ul, unsigned char>::push_back(NEO::ClDevice*&&)' at /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/shared/source/utilities/stackvec.h:233:10,
    inlined from 'NEO::BuiltinDispatchInfoBuilder::populate<char const*&, NEO::MultiDeviceKernel*&>(unsigned int, NEO::ConstStringRef, char const*&, NEO::MultiDeviceKernel*&)void' at /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/opencl/source/built_ins/populate_built_ins.inl:15:27:
/usr/include/c++/14.2.0/bits/vector.tcc:114:27: error: array subscript 'struct vector[0]' is partly outside array bounds of 'struct ClDeviceVector[1]' [-Werror=array-bounds=]
  114 |         if (this->_M_impl._M_finish != this->_M_impl._M_end_of_storage)
      |             ~~~~~~~~~~~~~~^~~~~~~~~
In file included from /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/opencl/source/built_ins/vme_builtin.cpp:15:
[... cut ...]
/builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.35.30872.22/opencl/source/built_ins/populate_built_ins.inl:14:20: note: at offset 8 into object 'deviceVector' of size 24
   14 |     ClDeviceVector deviceVector;
      |                    ^~~~~~~~~~~~
cc1plus: all warnings being treated as errors
ninja: subcommands failed
```

Tried with 24.35.30872.22 and the following CMake config

```

	cmake -B build -G Ninja \
		-DCMAKE_BUILD_TYPE=None \
		-DCMAKE_INSTALL_PREFIX=/usr \
		-DCMAKE_INSTALL_LIBDIR=lib \
		-DSUPPORT_DG1=ON \
		-DSUPPORT_DG2=ON \
		-DDISABLE_WDDM_LINUX=ON \
		-DNEO_DISABLE_LTO=OFF \
		-DNEO_SKIP_UNIT_TESTS=1 \
		.
```

JablonskiMateusz, 2024-10-22:
@bl4ckb0ne  what changed since these comments were added https://github.com/intel/compute-runtime/issues/763#issuecomment-2344012437 https://github.com/intel/compute-runtime/issues/763#issuecomment-2353783192?

bl4ckb0ne, 2024-10-22:
In https://github.com/intel/compute-runtime/issues/763#issuecomment-2353783192 I manually edited the header on my system to build it locally
In https://github.com/intel/compute-runtime/issues/763#issuecomment-2344012437 I managed to get the alpinelinux CI to build the project and produce an installable package without modifying the existing system

JablonskiMateusz, 2024-10-22:
@bl4ckb0ne so it seems like the gmmlib patch fixes the initial issue? 
Regarding the new issue, could you re-test on the recent release 24.39?

bl4ckb0ne, 2024-10-22:
Yep, the patch fixes the first issue.

From the CI with the latest 24.39 release

```
ninja: job failed: /usr/bin/g++ @shared/source/CMakeFiles/neo_shared.dir/os_interface/linux/drm_memory_manager_create_multi_host_allocation.cpp.o.rsp -MD -MT shared/source/CMakeFiles/neo_shared.dir/os_interface/linux/drm_memory_manager_create_multi_host_allocation.cpp.o -MF shared/source/CMakeFiles/neo_shared.dir/os_interface/linux/drm_memory_manager_create_multi_host_allocation.cpp.o.d -o shared/source/CMakeFiles/neo_shared.dir/os_interface/linux/drm_memory_manager_create_multi_host_allocation.cpp.o -c /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/shared/source/os_interface/linux/drm_memory_manager_create_multi_host_allocation.cpp
In file included from /usr/include/c++/14.2.0/vector:66,
                 from /usr/include/c++/14.2.0/functional:64,
                 from /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/shared/source/helpers/aligned_memory.h:15,
                 from /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/shared/source/os_interface/linux/drm_memory_manager_create_multi_host_allocation.cpp:9:
In member function 'std::vector<NEO::BufferObject*, std::allocator<NEO::BufferObject*> >::push_back(NEO::BufferObject* const&)',
    inlined from 'StackVec<NEO::BufferObject*, 4ul, unsigned char>::push_back(NEO::BufferObject* const&)' at /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/shared/source/utilities/stackvec.h:225:34,
    inlined from 'StackVec<NEO::BufferObject*, 4ul, unsigned char>::push_back(NEO::BufferObject* const&)' at /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/shared/source/utilities/stackvec.h:221:10,
    inlined from 'NEO::DrmAllocation::DrmAllocation(unsigned int, unsigned long, NEO::AllocationType, NEO::BufferObject*, void*, unsigned long, unsigned long, NEO::MemoryPool)' at /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/shared/source/os_interface/linux/drm_allocation.h:48:32:
/usr/include/c++/14.2.0/bits/stl_vector.h:1285:27: error: 'MEM[(struct vector *)this_10(D) + 1528B].D.185608._M_impl.D.184944._M_finish' may be used uninitialized [-Werror=maybe-uninitialized]
 1285 |         if (this->_M_impl._M_finish != this->_M_impl._M_end_of_storage)
      |             ~~~~~~~~~~~~~~^~~~~~~~~
/usr/include/c++/14.2.0/bits/stl_vector.h:1285:54: error: 'MEM[(struct vector *)this_10(D) + 1528B].D.185608._M_impl.D.184944._M_end_of_storage' may be used uninitialized [-Werror=maybe-uninitialized]
 1285 |         if (this->_M_impl._M_finish != this->_M_impl._M_end_of_storage)
      |                                        ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
```

JablonskiMateusz, 2024-11-04:
@bl4ckb0ne  could you share more details on used compiler  and used compile options? 
I don't see how the vector could be uninitialized, it is allocated within `ensureDynamicMem` and then `usesDynamicMem` returns success

bl4ckb0ne, 2024-11-06:
I'm building with GCC 14.2.0 with the `Release` CMake build type.

bl4ckb0ne, 2024-11-06:
Fails early on, probably something I can patch locally

```
[896/4059] Building CXX object offline_compiler/source/CMakeFiles/ocloc_lib.dir/ocloc_dll_options.cpp.o
ninja: job failed: /usr/bin/clang++ @offline_compiler/source/CMakeFiles/ocloc_lib.dir/__/__/source/os_interface/linux/sys_calls_linux.cpp.o.rsp -MD -MT offline_compiler/source/CMakeFiles/ocloc_lib.dir/__/__/source/os_interface/linux/sys_calls_linux.cpp.o -MF offline_compiler/source/CMakeFiles/ocloc_lib.dir/__/__/source/os_interface/linux/sys_calls_linux.cpp.o.d -o offline_compiler/source/CMakeFiles/ocloc_lib.dir/__/__/source/os_interface/linux/sys_calls_linux.cpp.o -c /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/shared/source/os_interface/linux/sys_calls_linux.cpp
/builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/shared/source/os_interface/linux/sys_calls_linux.cpp:69:36: error: implicit conversion loses integer precision: 'unsigned long' to 'int' [-Werror,-Wshorten-64-to-32]
   69 |     return ::ioctl(fileDescriptor, request, arg);
      |            ~~                      ^~~~~~~
1 error generated.
ninja: subcommand failed
```

bl4ckb0ne, 2024-11-11:
ICE with clang

```
inja: job failed: cd /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/shared/source/built_ins/kernels && /usr/bin/cmake -E env LD_LIBRARY_PATH=:/builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/ocloc-24.39.1 -q -file /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/built_ins/x64/xe2_hpg_core/stateless_copy_buffer_rect_stateless_lnl.spv -spirv_input -device 20.4.4 -cl-intel-greater-than-4GB-buffer-required -64 -stateful_address_mode bindful -output stateless_copy_buffer_rect_stateless_20_4_4 -output_no_suffix -out_dir /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/built_ins/x64/xe2_hpg_core -options -cl-kernel-arg-info
error: IGC: 
internal compiler error
Build failed with error code: -11
Command was: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/ocloc-24.39.1 -q -file /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/built_ins/x64/xe2_hpg_core/stateless_copy_buffer_rect_stateless_lnl.spv -spirv_input -device 20.4.4 -cl-intel-greater-than-4GB-buffer-required -64 -stateful_address_mode bindful -output stateless_copy_buffer_rect_stateless_20_4_4 -output_no_suffix -out_dir /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/built_ins/x64/xe2_hpg_core -options "-cl-kernel-arg-info"
ninja: job failed: cd /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/shared/source/built_ins/kernels && /usr/bin/cmake -E env LD_LIBRARY_PATH=:/builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/ocloc-24.39.1 -q -file /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/built_ins/x64/xe2_hpg_core/bindless_copy_image_to_image1d_lnl.spv -spirv_input -device 20.4.4 -force_stos_opt -64 -stateful_address_mode bindless -output bindless_copy_image_to_image1d_20_4_4 -output_no_suffix -out_dir /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/built_ins/x64/xe2_hpg_core -internal_options -cl-intel-use-bindless-mode\ -cl-intel-use-bindless-advanced-mode -options -cl-kernel-arg-info
error: IGC: 
internal compiler error
Build failed with error code: -11
Command was: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/ocloc-24.39.1 -q -file /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/built_ins/x64/xe2_hpg_core/bindless_copy_image_to_image1d_lnl.spv -spirv_input -device 20.4.4 -force_stos_opt -64 -stateful_address_mode bindless -output bindless_copy_image_to_image1d_20_4_4 -output_no_suffix -out_dir /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/built_ins/x64/xe2_hpg_core -internal_options "-cl-intel-use-bindless-mode -cl-intel-use-bindless-advanced-mode" -options "-cl-kernel-arg-info"
ninja: job failed: cd /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/shared/source/built_ins/kernels && /usr/bin/cmake -E env LD_LIBRARY_PATH=:/builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/ocloc-24.39.1 -q -file /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/built_ins/x64/xe2_hpg_core/bindful_fill_buffer_lnl.spv -spirv_input -device 20.4.4 -force_stos_opt -64 -stateful_address_mode bindful -output bindful_fill_buffer_20_4_4 -output_no_suffix -out_dir /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/built_ins/x64/xe2_hpg_core -options -cl-kernel-arg-info
error: IGC: 
internal compiler error
Build failed with error code: -11
Command was: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/ocloc-24.39.1 -q -file /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/built_ins/x64/xe2_hpg_core/bindful_fill_buffer_lnl.spv -spirv_input -device 20.4.4 -force_stos_opt -64 -stateful_address_mode bindful -output bindful_fill_buffer_20_4_4 -output_no_suffix -out_dir /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/built_ins/x64/xe2_hpg_core -options "-cl-kernel-arg-info"
ninja: job failed: cd /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/shared/source/built_ins/kernels && /usr/bin/cmake -E env LD_LIBRARY_PATH=:/builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/ocloc-24.39.1 -q -file /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/built_ins/x64/xe2_hpg_core/bindless_fill_image3d_bmg.spv -spirv_input -device 20.1.4 -force_stos_opt -64 -stateful_address_mode bindless -output bindless_fill_image3d_20_1_4 -output_no_suffix -out_dir /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/ [... cut ...] build/bin/built_ins/x64/xe2_hpg_core/stateless_copy_buffer_to_image3d_stateless_lnl.spv -spirv_input -device 20.4.4 -cl-intel-greater-than-4GB-buffer-required -64 -stateful_address_mode bindful -output stateless_copy_buffer_to_image3d_stateless_20_4_4 -output_no_suffix -out_dir /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.39.31294.12/build/bin/built_ins/x64/xe2_hpg_core -options "-cl-kernel-arg-info"
ninja: subcommands failed
>>> ERROR: intel-opencl-icd: build failed
```

eero-t, 2025-02-12:
> ICE with clang

@bl4ckb0ne That's (Alpine?) clang bug, not compute-runtime one.

bl4ckb0ne, 2025-02-12:
Yes alpine clang. Since its building fine I'll close the issue. Thanks everybody.

closed 2025-02-12 by bl4ckb0ne