intel/compute-runtime #770: The latest `intel-igc-{core,opencl}` cannot coexist with legacy packages

nyanmisaka, 2024-10-21:
The [legacy ICD](https://github.com/intel/compute-runtime/blob/master/LEGACY_PLATFORMS.md) requires (intel-igc-opencl = 1.0.17537.20), therefore, it is not possible to install both [new](https://github.com/intel/compute-runtime/releases/tag/24.39.31294.12) and legacy `compute-runtime` in the same environment.

Thus, if our software is to support both platforms, we will have to stay on version [24.35.30872.22](https://github.com/intel/compute-runtime/releases/tag/24.35.30872.22). This will cause problems when new platforms are introduced in the future, so please release a new version to resolve this version conflict.

```console
sudo apt install ./*.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'intel-igc-core' instead of './intel-igc-core_1.0.17791.9_amd64.deb'
Note, selecting 'intel-igc-opencl' instead of './intel-igc-opencl_1.0.17791.9_amd64.deb'
Note, selecting 'intel-level-zero-gpu' instead of './intel-level-zero-gpu_1.6.31294.12_amd64.deb'
Note, selecting 'intel-opencl-icd' instead of './intel-opencl-icd_24.39.31294.12_amd64.deb'
Note, selecting 'libigdgmm12' instead of './libigdgmm12_22.5.2_amd64.deb'
The following packages will be REMOVED:
  intel-level-zero-gpu-legacy1 intel-opencl-icd-legacy1
The following packages will be upgraded:
  intel-igc-core intel-igc-opencl intel-level-zero-gpu intel-opencl-icd libigdgmm12
5 upgraded, 0 newly installed, 2 to remove and 0 not upgraded.
Need to get 0 B/156 MB of archives.
After this operation, 14.9 MB disk space will be freed.
Do you want to continue? [Y/n]
```

```console
apt show intel-opencl-icd-legacy1
Package: intel-opencl-icd-legacy1
Version: 24.35.30872.22
Status: install ok installed
Priority: optional
Section: libs
Source: intel-compute-runtime
Maintainer: Intel Graphics Team <linux-graphics@intel.com>
Original-Maintainer: Debian OpenCL Maintainers <pkg-opencl-devel@lists.alioth.debian.org>
Installed-Size: 10.8 MB
Provides: opencl-icd
Depends: libc6 (>= 2.34), libgcc-s1 (>= 3.4), libigdgmm12 (>= 22.5.0), libstdc++6 (>= 12), intel-igc-opencl (= 1.0.17537.20), ocl-icd-libopencl1
Breaks: intel-opencl
Replaces: intel-opencl
Homepage: https://github.com/intel/compute-runtime
Download-Size: unknown
APT-Manual-Installed: yes
APT-Sources: /var/lib/dpkg/status
Description: Intel graphics compute runtime for OpenCL
 The Intel(R) Graphics Compute Runtime for OpenCL(TM) is a open source
 project to converge Intel's development efforts on OpenCL(TM) compute
 stacks supporting the GEN graphics hardware architecture.
```

nyanmisaka, 2024-10-22:
@JablonskiMateusz One more question, Rocket Lake (UHD7xx) has the same Gen12 Gfx as Tiger Lake but is excluded from the latest supported list. Is it a typo?

JablonskiMateusz, 2024-10-22:
> @JablonskiMateusz One more question, Rocket Lake (UHD7xx) has the same Gen12 Gfx as Tiger Lake but is excluded from the latest supported list. Is it a typo?

Yes, that's a typo
@kmachnik  FYI

kmachnik, 2024-10-22:
Typo corrected. Rocket Lake again present in the supported platforms table.
Thank you for catching that!

kmachnik, 2024-12-03:
To get legacy packages from 24.35 working together with 24.45 release, follow the steps below:

- start with a clean install of 24.35
- install driver 24.45 on top of it 
- to resolve any inconsistencies, use:

```
apt-get install --fix-broken 
```

nyanmisaka, 2024-12-04:
@kmachnik
Thanks for the update! I can confirm that `24.45.31740.9` and `24.35.30872.22`(legacy) can coexist on Ubuntu 24.04.

But I want to let you know that a new issue has appeared, that the prebuilt `libigc2` packages no longer support Ubuntu 22.04, while the IGC [README](https://github.com/intel/intel-graphics-compiler/blob/master/README.md#supported-linux-versions) says that both 22.04 and 24.04 are supported. 

```
The following packages have unmet dependencies:
 intel-igc-core-2 : Depends: libc6 (>= 2.38) but 2.35-0ubuntu3.8 is to be installed
 intel-igc-opencl-2 : Depends: libc6 (>= 2.38) but 2.35-0ubuntu3.8 is to be installed
```

This is because the packages built on 24.04 require a higher `glibc` version, which makes them incompatible with backwards. Is there any chance to fix this as well? Many users are still on 22.04 LTS.

GermanAizek, 2024-12-18:
@kmachnik,
Debian Testing Linux debian 6.12.5-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.12.5-1 (2024-12-15) x86_64 GNU/Linux

```
$ sudo apt install intel-opencl-icd
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

Unsatisfied dependencies:
 intel-opencl-icd : Depends: libigc1 (< 1.0.17537.20+~) but 1.0.17791.18-1 is to be installed
                    Depends: libigdfcl1 (< 1.0.17537.20+~) but 1.0.17791.18-1 is to be installed
Error: Unable to correct problems, you have held broken packages.
```

nyanmisaka, 2024-12-18:
@GermanAizek The packages provided in this repo **replace** the ones provided by Debian/Ubuntu, they will not be compatible with each other. So please don't hijack this thread as they are different things.

```
wget https://github.com/intel/compute-runtime/releases/download/24.45.31740.9/intel-opencl-icd_24.45.31740.9_amd64.deb

dpkg -I ./intel-opencl-icd_24.45.31740.9_amd64.deb

 new Debian package, version 2.0.
 size 2464734 bytes: control archive=1068 bytes.
      30 bytes,     1 lines      conffiles
     816 bytes,    18 lines      control
     423 bytes,     6 lines      md5sums
     194 bytes,    12 lines   *  postinst             #!/bin/sh
     203 bytes,    14 lines   *  prerm                #!/bin/sh
 Package: intel-opencl-icd
 Source: intel-compute-runtime
 Version: 24.45.31740.9
 Architecture: amd64
 Maintainer: Intel Graphics Team <linux-graphics@intel.com>
 Installed-Size: 21271
 Depends: libc6 (>= 2.34), libgcc-s1 (>= 3.4), libigdgmm12 (>= 22.5.2), libstdc++6 (>= 12), intel-igc-opencl-2 (= 2.1.12), ocl-icd-libopencl1
 Breaks: intel-opencl
 Replaces: intel-opencl
 Provides: opencl-icd
 Section: libs
 Priority: optional
 Homepage: https://github.com/intel/compute-runtime
 Description: Intel graphics compute runtime for OpenCL
  The Intel(R) Graphics Compute Runtime for OpenCL(TM) is a open source
  project to converge Intel's development efforts on OpenCL(TM) compute
  stacks supporting the GEN graphics hardware architecture.
 Original-Maintainer: Debian OpenCL Maintainers <pkg-opencl-devel@lists.alioth.debian.org>
```

GermanAizek, 2024-12-18:
@nyanmisaka,
yes, I deleted all APT intel-opencl* and depends Debian packages as well as from ppa:intel and manually installed them from Github Releases. And it helped.

nyanmisaka, 2024-12-24:
```
intel-igc-core-2/now 2.2.3 amd64 [installed,local]
intel-igc-core/now 1.0.17537.20 amd64 [installed,local]
intel-igc-opencl-2/now 2.2.3 amd64 [installed,local]
intel-igc-opencl/now 1.0.17537.20 amd64 [installed,local]
intel-level-zero-gpu-legacy1/now 1.3.30872.22 amd64 [installed,local]
intel-level-zero-gpu/now 1.6.31907.7 amd64 [installed,local]
intel-opencl-icd-legacy1/now 24.35.30872.22 amd64 [installed,local]
intel-opencl-icd/now 24.48.31907.7 amd64 [installed,local]
```

Closing in https://github.com/intel/compute-runtime/releases/tag/24.48.31907.7

closed 2024-12-24 by nyanmisaka