intel/compute-runtime #773: Recent Intel compute runtime driver update doesnt detect igpu

SkyLLerGithub, 2024-11-02:
OS: Arch Linux x86
CPU: intel i5 10210 u
GPU: Intel UHD Graphics 620

Recent driver update doesnt detect igpu. Verified by running clinfo
[image]

Downgrading works and device is detected in clinfo.
(here is version 24.35.30872.22-2)
[image]

nyanmisaka, 2024-11-03:
https://gitlab.archlinux.org/archlinux/packaging/packages/intel-compute-runtime/-/issues/3

There's an issue ticket.

SkyLLerGithub, 2024-11-03:
> https://github.com/intel/compute-runtime/blob/master/LEGACY_PLATFORMS.md
> 
> It would be best to file an issue on Archlinux as well to ask the maintainers to separate current and legacy platforms into two different packages.

Thank you.

closed 2024-11-03 by SkyLLerGithub