intel/compute-runtime #774: Segfault when building the following kernel

bjourne, 2024-11-04:
Compiling and building the following OpenCL program causes the latest version of the OpenCL runtime to crash:
```
typedef struct {
    uint n_mails;
    uint mails[100];
} mailbox;

__kernel void
run() {
    uint me = get_local_id(0);
    __local mailbox boxes[1][4];
    for (uint ix = 0; ix < 1; ix++) {
        __local mailbox *mb = &boxes[ix][me];
        uint i = 0;
        while (true) {
            uint swap = (i * 2) + 1;
            if (swap >= mb->n_mails) {
                break;
            }
            mb->mails[i] = mb->mails[swap];
            i = swap;
        }
    }
}
```
Compilation crashes on the following device:
```
  Name           : Intel(R) Xeon(R) W-2245 CPU @ 3.90GHz
  Version        : OpenCL 3.0 (Build 0)
  Driver         : 2024.18.7.0.11_160000
  C Version      : OpenCL C 3.0 
  Compute units  : 16 
  Global memory  : 33.33 GB
  Max allocation : 16.66 GB
  Max wg. size   : 8 k
  Local mem. size: 32.77 kB
  Max work items : 8192:8192:8192
  Pipe support   : yes
  Compiler usable: yes
  Image support  : yes
```
But works fine on the following device:
```
  Name           : Intel(R) Xeon(R) W-2245 CPU @ 3.90GHz
  Version        : OpenCL 3.0 (Build 0)
  Driver         : 2024.17.3.0.08_160000
  C Version      : OpenCL C 3.0 
  Compute units  : 16 
  Global memory  : 33.33 GB
  Max allocation : 16.66 GB
  Max wg. size   : 8 k
  Local mem. size: 32.77 kB
  Max work items : 8192:8192:8192
  Pipe support   : yes
  Compiler usable: yes
  Image support  : yes
```
So I guess a bug was introduced relatively recently.

This OpenCL program causes the compiler to crash when building:

```
typedef struct {
    uint n_mails;
    uint mails[100];
} mailbox;

uint
mailbox_take(__local mailbox *me) {
    me->n_mails--;
    uint temp = me->mails[me->n_mails];
    uint min = me->mails[0];
    uint i = 0;
    while (true) {
        uint swap = (i * 2) + 1;
        if (swap >= me->n_mails) {
            break;
        }
        uint other = swap + 1;
        if ((other < me->n_mails) &&
            (me->mails[other] <= me->mails[swap])) {
            swap = other;
        }
        if (temp <= me->mails[swap]) {
            break;
        }
        me->mails[i] = me->mails[swap];
        i = swap;
    }
    me->mails[i] = temp;
    return min;
}

__kernel void
run(const uint T, const uint N, __global uint *mails) {
    uint me = get_local_id(0);
    __local mailbox boxes[4][4];
    for (uint t = 0; t < T; t++) {
        for (uint i = 0; i < 4; i++) {
            __local mailbox *box = &boxes[i][me];
            while(box->n_mails > 0 && box->mails[0] == t) {
                uint m = mailbox_take(box);
            }
        }
    }
}
```

Here is my OpenCL platform:

  Name           : Intel(R) Xeon(R) W-2245 CPU @ 3.90GHz
  Version        : OpenCL 3.0 (Build 0)
  Driver         : 2024.18.7.0.11_160000
  C Version      : OpenCL C 3.0 
  Compute units  : 16 
  Global memory  : 33.33 GB
  Max allocation : 16.66 GB
  Max wg. size   : 8 k
  Local mem. size: 32.77 kB
  Max work items : 8192:8192:8192
  Pipe support   : yes
  Compiler usable: yes
  Image support  : yes

JablonskiMateusz, 2024-11-07:
@bjourne  Please duplicate the issue to https://github.com/intel/intel-graphics-compiler repo

cdai2, 2026-01-12:
This failure was already fixed since OCL CPU RT 2025 / 4 /22 release.