intel/compute-runtime #776: ZE_RESULT_ERROR_DEPENDENCY_UNAVAILABLE while running sample level zero program and intel extension for openxla

shibdas, 2024-11-12:
Hi,
  I was trying to run a sample JAX python program with intel extension for openxla on an Intel flex 140 GPU  when I encountered an error ZE_RESULT_ERROR_DEPENDENCY_UNAVAILABLE  coming out of zeModuleCreate(). I then tried to run a sample level zero program from https://github.com/jjfumero/codeBlogArticles/blob/master/june2021/levelZero_MxM/mxm.cpp and it also returned the same error code at https://github.com/jjfumero/codeBlogArticles/blob/fc9b5bc4fee36c1b3488c3d0ebef5c690a138822/june2021/levelZero_MxM/mxm.cpp#L150

        auto status = zeModuleCreate(context, device, &moduleDesc, &module, &buildLog);
        if (status != ZE_RESULT_SUCCESS) {

I believe this is happening due to some mismatched compiler and runtime but I am not able to figure out what is missing here. I have oneapi 2024.2 installed.

clinfo -l
Platform #0: Intel(R) OpenCL
 `-- Device #0: Intel(R) Xeon(R) Silver 4314 CPU @ 2.40GHz
Platform #1: Intel(R) OpenCL Graphics
 `-- Device #0: Intel(R) Data Center GPU Flex 140

dpkg -l | grep "intel"
ii  intel-basekit                                  2024.2.1-98                             amd64        Intel® oneAPI Base Toolkit
ii  intel-basekit-env-2024.2                       2024.2.1-98                             all          Intel® oneAPI Base Toolkit
ii  intel-basekit-getting-started-2024.2           2024.2.1-98                             all          Intel® oneAPI Base Toolkit
ii  intel-fw-gpu                                   2023.39.2-255~22.04                     all          Firmware package for Intel integrated and discrete GPUs
ii  intel-gpu-tools                                1.26-2                                  amd64        tools for debugging the Intel graphics driver
ii  intel-gsc                                      0.8.9+65~u22.04                         amd64        Intel(R) Graphics System Controller Firmware
ii  intel-gsc-dev                                  0.8.16+88~u22.04                        amd64        Intel(R) Graphics System Controller Firmware
ii  intel-igc-cm                                   1.0.202-682~22.04                       amd64        Intel(R) C for Metal Compiler -- CM Frontend lib
ii  intel-igc-core                                 1.0.12149.1                             amd64        Intel(R) Graphics Compiler for OpenCL(TM)
ii  intel-igc-opencl                               1.0.12149.1                             amd64        Intel(R) Graphics Compiler for OpenCL(TM)
ii  intel-level-zero-gpu                           1.3.29735.27-914~22.04                  amd64        Intel(R) Graphics Compute Runtime for oneAPI Level Zero.
ii  intel-level-zero-gpu-dev                       1.3.29735.27-914~22.04                  amd64        Intel(R) Graphics Compute Runtime for oneAPI Level Zero
ii  intel-media-va-driver-non-free:amd64           23.2.4-682~22.04                        amd64        VAAPI driver for the Intel GEN8+ Graphics family
ii  intel-metrics-discovery                        1.12.166-682~22.04                      amd64        Intel(R) Metrics Discovery Application Programming Interface --
ii  intel-metrics-library                          1.0.133-649~22.04                       amd64        Intel(R) Metrics Library for MDAPI (Intel(R) Metrics Discovery
ii  intel-microcode                                3.20240910.0ubuntu0.22.04.1             amd64        Processor microcode firmware for Intel CPUs
ii  intel-ocloc                                    24.39.31294.20-1032~22.04               amd64        Tool for managing Intel Compute GPU device binary format
ii  intel-oneapi-advisor                           2024.3.0-39                             amd64        Intel® Advisor
ii  intel-oneapi-ccl-2021.13                       2021.13.1-31                            amd64        Intel® oneAPI Collective Communications Library Runtime Environment
ii  intel-oneapi-ccl-devel                         2021.13.1-31                            amd64        Intel® oneAPI Collective Communications Library
ii  intel-oneapi-ccl-devel-2021.13                 2021.13.1-31                            amd64        Intel® oneAPI Collective Communications Library
ii  intel-oneapi-common-licensing                  2024.2.1-21                             all          oneAPI Common License
ii  intel-oneapi-common-licensing-2024.2           2024.2.1-21                             all          oneAPI Common License
ii  intel-oneapi-common-oneapi-vars                2024.2.1-21                             all          oneAPI Common Toolkit Environment Script
ii  intel-oneapi-common-oneapi-vars-2024.2         2024.2.1-21                             all          oneAPI Common Toolkit Environment Script
ii  intel-oneapi-common-vars                       2024.2.1-21                             all          oneAPI Common Environment Scripts
ii  intel-oneapi-compiler-cpp-eclipse-cfg-2024.2   2024.2.1-1079                           all          Intel® oneAPI DPC++/C++ Compiler 2024.2.1 for Linux* eclipse integration configuration file (C++)
ii  intel-oneapi-compiler-dpcpp-cpp                2024.2.1-1079                           amd64        Intel® oneAPI DPC++/C++ Compiler
ii  intel-oneapi-compiler-dpcpp-cpp-2024.2         2024.2.1-1079                           amd64        Intel® oneAPI DPC++/C++ Compiler
ii  intel-oneapi-compiler-dpcpp-cpp-common-2024.2  2024.2.1-1079                           all          Intel® oneAPI DPC++/C++ Compiler 2024.2.1 for Linux*
ii  intel-oneapi-compiler-dpcpp-cpp-runtime-2024.2 2024.2.1-1079                           amd64        Intel® oneAPI DPC++/C++ Compiler 2024.2.1 for Linux* runtime package for Intel(R) 64
ii  intel-oneapi-compiler-dpcpp-eclipse-cfg-2024.2 2024.2.1-1079                           all          Intel® oneAPI DPC++/C++ Compiler 2024.2.1 for Linux* eclipse integration configuration file (DPC++)
ii  intel-oneapi-compiler-shared-2024.2            2024.2.1-1079                           amd64        Intel(R) Compiler Shared Files
ii  intel-oneapi-compiler-shared-common-2024.2     2024.2.1-1079                           all          Intel(R) Compiler Shared Files
ii  intel-oneapi-compiler-shared-runtime-2024.2    2024.2.1-1079                           amd64        Intel(R) Compiler Shared Files runtime contents
ii  intel-oneapi-dal-2024.7                        2024.7.0-13                             amd64        Intel® oneAPI Data Analytics Library
ii  intel-oneapi-dal-common-2024.7                 2024.7.0-13                             all          Intel® oneAPI Data Analytics Library common
ii  intel-oneapi-dal-common-devel-2024.7           2024.7.0-13                             all          Intel® oneAPI Data Analytics Library common
ii  intel-oneapi-dal-devel                         2024.7.0-13                             amd64        Intel® oneAPI Data Analytics Library Development Package
ii  intel-oneapi-dal-devel-2024.7                  2024.7.0-13                             amd64        Intel® oneAPI Data Analytics Library Development Package
ii  intel-oneapi-dev-utilities                     2024.2.1-11                             amd64        Dev Utilities
ii  intel-oneapi-dev-utilities-2024.2              2024.2.1-11                             amd64        Dev Utilities
ii  intel-oneapi-dev-utilities-eclipse-cfg-2024.2  2024.2.1-11                             all          intel-oneapi-dev-utilities-eclipse-cfg
ii  intel-oneapi-diagnostics-utility               2024.2.1-13                             amd64        Diagnostics Utility for Intel® oneAPI Toolkits
ii  intel-oneapi-diagnostics-utility-2024.2        2024.2.1-13                             amd64        Diagnostics Utility for Intel® oneAPI Toolkits
ii  intel-oneapi-dnnl                              2024.2.1-74                             amd64        Intel® oneAPI Deep Neural Network Library
ii  intel-oneapi-dnnl-2024.2                       2024.2.1-74                             amd64        Intel® oneAPI Deep Neural Network Library
ii  intel-oneapi-dnnl-devel                        2024.2.1-74                             amd64        Intel® oneAPI Deep Neural Network Library Development Package
ii  intel-oneapi-dnnl-devel-2024.2                 2024.2.1-74                             amd64        Intel® oneAPI Deep Neural Network Library Development Package
[... cut ...]
ii  libdrm-intel1:amd64                            2.4.113-2~ubuntu0.22.04.1               amd64        Userspace interface to intel-specific kernel DRM services -- runtime
ii  libopenvino-intel-cpu-plugin-2023.1.0          2023.1.0.12185                          amd64        OpenVINO™ Toolkit
ii  libopenvino-intel-gpu-plugin-2023.1.0          2023.1.0.12185                          amd64        OpenVINO™ Toolkit
ii  xserver-xorg-video-intel                       2:2.99.917+git20210115-1                amd64        X.Org X server -- Intel i8xx, i9xx display driver

JablonskiMateusz, 2024-11-13:
Hi @shibdas 
Please capture strace logs

JablonskiMateusz, 2024-11-19:
it looks like some compiler incompatibility...
Please run the app with additional env:
`NEOReadDebugKeys=1 PrintDebugMessages=1`

shibdas, 2024-11-19:
./mxm
WARNING: Failed to request OCL Turbo Boost
NEO_CACHE_PERSISTENT is enabled. Cache is located in: /home/test/.cache/neo_compiler_cache

Installed Compiler Library libigc.so.1 is incompatible
DeviceBinaryFormat::zebin::.ze_info : Minor version : 41 is newer than available in decoder : 39 - some features may be skipped

DeviceBinaryFormat::zebin : Unhandled SHT_NOTE section : .note.intelgt.metrics currently supports only : .note.intelgt.compat.
DeviceBinaryFormat::zebin::.ze_info : Minor version : 41 is newer than available in decoder : 39 - some features may be skipped

computeUnits for each thread: 2048
perHwThreadPrivateMemorySize: 0	 totalPrivateMemorySize: 0
perHwThreadScratchSize: 0	 totalScratchSize: 0
perHwThreadPrivateScratchSize: 0	 totalPrivateScratchSize: 0
Device   : Intel(R) Data Center GPU Flex 140
Type     : GPU
Vendor ID: 8086
#Queue Groups: 2
NEO_CACHE_PERSISTENT is enabled. Cache is located in: /home/test/.cache/neo_compiler_cache

Installed Compiler Library libigc.so.1 is incompatible
Build log: 
Segmentation fault (core dumped)

Do I need to install a specific libigc package? This is what I have

ii  libigc-dev                                     1.0.17537.24-996~22.04                  amd64        Intel graphics compiler for OpenCL -- core development files
ii  libigc1                                        1.0.17537.24-996~22.04                  amd64        Intel graphics compiler for OpenCL -- core libs

JablonskiMateusz, 2024-11-25:
Hi @shibdas  
It looks like you have conflicting packages on the system.
Currently we cannot establish compatibility between our driver and compiler and therefore we have strict dependency for given version. This also makes that if you want to have OCL and L0 runtimes, you should take them from the same version as they both will require the same version of IGC. 

However, I see there is completely mess with the versions on the system.
> ii intel-igc-core 1.0.12149.1 amd64 Intel(R) Graphics Compiler for OpenCL(TM)
ii intel-igc-opencl 1.0.12149.1 amd64 Intel(R) Graphics Compiler for OpenCL(TM)

this is very old igc, corresponding to our 22.38 release

>ii intel-level-zero-gpu 1.3.29735.27-91422.04 amd64 Intel(R) Graphics Compute Runtime for oneAPI Level Zero.
ii intel-level-zero-gpu-dev 1.3.29735.27-91422.04 amd64 Intel(R) Graphics Compute Runtime for oneAPI Level Zero

this level zero runtime is also quite old, corresponds to 24.22 release

>ii intel-ocloc 24.39.31294.20-103222.04 amd64 Tool for managing Intel Compute GPU device binary format

ocloc corresponds to our latest, 24.39 release

> ii intel-opencl-icd 24.39.31294.20-103222.04 amd64 Intel graphics compute runtime for OpenCL

OCL runtime also corresponds to our latest, 24.39 release

>ii libigc-dev 1.0.17537.24-99622.04 amd64 Intel graphics compiler for OpenCL -- core development files
ii libigc1 1.0.17537.24-99622.04 amd64 Intel graphics compiler for OpenCL -- core libs

These IGC libraries corresponds to our 24.35 release 

Could you please adjust the packages to the same release base and then retry?

shibdas, 2024-11-25:
Can you please point me to a guide  install all the packages properly after
I uninstall all of them?

pwilma, 2025-10-08:
Hi @shibdas ,
We’d like to know if this issue is still affecting you. If so, please provide an update or any additional information. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated!

closed 2025-11-19 by JablonskiMateusz