intel/compute-runtime #779: ZE_RESULT_ERROR_UNINITIALIZED when test with zeInit(0) on MTL

LoSealL, 2024-12-04:
**Test OS**: Ubuntu 20.04
**Kernel**: 5.15.0-126-generic
**CPU**: Core Ultra 5 135H

- [x] build from source

Run test `zello_world_gpu`, it fails at `zeInit(0)` with error `ZE_RESULT_ERROR_UNINITIALIZED`. It can be enabled when OS is **Ubuntu 22.04**, but fails when we downgraded to 20.04.

eero-t, 2024-12-18:
> It can be enabled when OS is **Ubuntu 22.04**, but fails when we downgraded to 20.04.

That's expected.  Ubuntu 20.4 kernel and compute driver versions:
* https://packages.ubuntu.com/source/focal/intel-compute-runtime
* https://github.com/intel/compute-runtime/releases/tag/20.13.16352

are from early 2020, and MTL was launched at the very end of 2023:
* https://en.wikipedia.org/wiki/Meteor_Lake

You need **both** kernel and user-space driver versions that support MTL i.e. something that are released this year.

If you have Ubuntu subscription, you could request them to include new compute driver to their HWE enabling driver versions for older LTS versions.

Or if you are comfortable in building your own kernel, you could build & install new enough ones for MTL support.

JablonskiMateusz, 2025-07-21:
@LoSealL Could you please let us know if the above response addressed your issue?

closed 2025-07-21 by LoSealL