intel/compute-runtime #780: WW45 release checksums do not match

RossBrunton, 2024-12-04:
For this release: https://github.com/intel/compute-runtime/releases/tag/24.45.31740.9

The following indicates a checksum failure:

```
wget https://github.com/intel/intel-graphics-compiler/releases/download/v2.1.12/intel-igc-core-2_2.1.12+18087_amd64.deb
wget https://github.com/intel/intel-graphics-compiler/releases/download/v2.1.12/intel-igc-opencl-2_2.1.12+18087_amd64.deb
wget https://github.com/intel/compute-runtime/releases/download/24.45.31740.9/intel-level-zero-gpu-dbgsym_1.6.31740.9_amd64.ddeb
wget https://github.com/intel/compute-runtime/releases/download/24.45.31740.9/intel-level-zero-gpu_1.6.31740.9_amd64.deb
wget https://github.com/intel/compute-runtime/releases/download/24.45.31740.9/intel-opencl-icd-dbgsym_24.45.31740.9_amd64.ddeb
wget https://github.com/intel/compute-runtime/releases/download/24.45.31740.9/intel-opencl-icd_24.45.31740.9_amd64.deb
wget https://github.com/intel/compute-runtime/releases/download/24.45.31740.9/libigdgmm12_22.5.2_amd64.deb
wget https://github.com/intel/compute-runtime/releases/download/24.45.31740.9/ww45.sum
sha256sum -c ww45.sum
```

For reference, the checksums listed on the release page (at the time of writing) are currently:
```
98fb380e4be8c16112d3cfbeb24bf4aaef73b0dceacb64fc8fc8c8ff98871bda intel-igc-core-2_2.1.12+18087_amd64.deb
967585a81d851e20af9f2dbe6e9333a67ff91a5a83638ff13bd0e191df53fd10 intel-igc-opencl-2_2.1.12+18087_amd64.deb
a4f2e43c7cc29af8787a6a8fbe6a79d5f9e66a99ce0c29c2af1cdd41df3ef65d intel-level-zero-gpu-dbgsym_1.6.31740.9_amd64.ddeb
fb2a6064f81af9e673f1508f4517ef2b510a09efb9407c0e46719ef6337466ae intel-level-zero-gpu_1.6.31740.9_amd64.deb
2134fda1cbc8c2c877dd9d456f062a94ddee5f060a10e2a596b52c5fb3a70a90 intel-opencl-icd-dbgsym_24.45.31740.9_amd64.ddeb
c7fc22ed2fb7fb17fa2a9a549a7b4c1b73d458bc34395b37ff75d55780115776 intel-opencl-icd_24.45.31740.9_amd64.deb
f9c8b8a81bc4b6bc5b3d4da9c82252d5c05c03d7917ae31c71aa4c4f579bad8f libigdgmm12_22.5.2_amd64.deb
```

However, the sha256s of the downloaded files are as follows:
```
e3c3fa990cd21b136df289ef3edd5158d2740d845f649071cbe8e9e47daab982  intel-igc-core-2_2.1.12+18087_amd64.deb
02fd309a6655d25267e1dba8036c88013533c7a06a1e04224ca35de39dc4859b  intel-igc-opencl-2_2.1.12+18087_amd64.deb
fb2a6064f81af9e673f1508f4517ef2b510a09efb9407c0e46719ef6337466ae  intel-level-zero-gpu_1.6.31740.9_amd64.deb
a4f2e43c7cc29af8787a6a8fbe6a79d5f9e66a99ce0c29c2af1cdd41df3ef65d  intel-level-zero-gpu-dbgsym_1.6.31740.9_amd64.ddeb
c7fc22ed2fb7fb17fa2a9a549a7b4c1b73d458bc34395b37ff75d55780115776  intel-opencl-icd_24.45.31740.9_amd64.deb
2134fda1cbc8c2c877dd9d456f062a94ddee5f060a10e2a596b52c5fb3a70a90  intel-opencl-icd-dbgsym_24.45.31740.9_amd64.ddeb
f9c8b8a81bc4b6bc5b3d4da9c82252d5c05c03d7917ae31c71aa4c4f579bad8f  libigdgmm12_22.5.2_amd64.deb
```

Note that only both igc files have mismatching checksums.

eero-t, 2024-12-18:
@JablonskiMateusz ?

kmachnik, 2024-12-19:
We consulted this situation with IGC and decided checksums for IGC will be published on IGC releases. It will allow us to avoid such issues. Both NEO and IGC releases were updated to address this inconsistency.

closed 2024-12-19 by JablonskiMateusz