intel/compute-runtime #781: Package Conflict When using Steam in Ubuntu

ca1ic0, 2024-12-05:
Recently , i install Ubuntu24.04 on my pc with A750 GPU,the detail of the PC is:
1.MB: Asrock B450
2.CPU:ryzen5500
3.GPU: A750
4. Memory:16GB DDR4 3400MHZ
After the install of OS, I install the computing package because i want to use oneAPI and ipex according to the official page 
https://dgpu-docs.intel.com/driver/client/overview.html
And i notice that this is a big change in the page , like package name and package need to be installed.
After the install of intel package , i install Steam,and launch Steam, Steam install some i386 package and it cause conflict in APT,as follows:
[image]

Then i try on Ubuntu24.04 and Ubuntu22.04, the conflict appear in both .
 
The command i execute
```bash
# Install the Intel graphics GPG public key wget -qO - https://repositories.intel.com/gpu/intel-graphics.key | \ sudo gpg --yes --dearmor --output /usr/share/keyrings/intel-graphics.gpg # Configure the repositories.intel.com package repository echo "deb [arch=amd64,i386 signed-by=/usr/share/keyrings/intel-graphics.gpg] https://repositories.intel.com/gpu/ubuntu noble client" | \ sudo tee /etc/apt/sources.list.d/intel-gpu-noble.list # Update the package repository meta-data sudo apt update # Install the compute-related packages apt-get install -y libze1 intel-level-zero-gpu intel-opencl-icd clinfo
```
And i also found that , if i install steam first and install intel gpu package later , the conflict disappear , but the steam and game wont work well .
[image]

After the contact with intel official consumer support , he reply  ： 
```
您好，这边进一步排查了您反馈的问题判断可能是一个运行库冲突的问题。对应的计算运行库的支持属于开发方面 ，建议您可以前往githu官方页面：https://github.com/intel/compute-runtime
进行问题的提交：https://github.com/intel/compute-runtime/issues
以获取进一步的帮助。
```
it is about that i should invoke a issue to report the problem!
Hope it could be solved soon , thank you !

JablonskiMateusz, 2024-12-09:
Hi @ca1ic0
Could you please share exact command lines to reproduce the issue?

JablonskiMateusz, 2024-12-10:
if you start from clean Ubuntu 24.04 and install only `libze1 intel-level-zero-gpu intel-opencl-icd clinfo` 
how do you conflict with `libigdgmm12` ? It is not installed automatically

ca1ic0, 2024-12-11:
> if you start from clean Ubuntu 24.04 and install only `libze1 intel-level-zero-gpu intel-opencl-icd clinfo` how do you conflict with `libigdgmm12` ? It is not installed automatically
sry ,i put the wrong picture because the problem happened  month ago.
[image]
later i'll re-install the OS ，and export a more detailed log and clear screenshot.

ca1ic0, 2024-12-17:
There is the terminal log
[terminal.log](https://github.com/user-attachments/files/18159105/terminal.log)
After install of intel package，install steam and launch it. I record the screen, there is the video

ca1ic0, 2024-12-17:
video 
https://github.com/user-attachments/assets/f7bda4d7-019c-461b-bd09-d5e2cccb605b

JablonskiMateusz, 2024-12-17:
It looks like some issue with mesa package, I don't see anything related to intel-compute-runtime in that log

ca1ic0, 2024-12-17:
> It looks like some issue with mesa package, I don't see anything related to intel-compute-runtime in that log
but if i dont install  libze1 intel-level-zero-gpu intel-opencl-icd clinfo ,  there wont be err in apt.
later i will re-install Ubuntu, and try install steam directly and record the log

ca1ic0, 2024-12-17:
Here is the video , no err this time
1. clean install the OS
2. install the Steam
3. launch Steam
https://github.com/user-attachments/assets/848c65f3-334f-45d8-a6e9-27d9bf6043ae

JablonskiMateusz, 2024-12-17:
> Here is the video , no err this time
> 
> 1. clean install the OS
> 2. install the Steam
> 3. launch Steam
>    https://github.com/user-attachments/assets/848c65f3-334f-45d8-a6e9-27d9bf6043ae

once you have this setup, please install `libze1 intel-level-zero-gpu intel-opencl-icd clinfo` and show results

ca1ic0, 2024-12-18:
> once you have this setup, please install `libze1 intel-level-zero-gpu intel-opencl-icd clinfo` and show results
1. clean install the OS
2. install `libze1 intel-level-zero-gpu intel-opencl-icd clinfo`  
3. install the Steam
4. launch Steam
behave is video
https://github.com/user-attachments/assets/f7bda4d7-019c-461b-bd09-d5e2cccb605b

ca1ic0, 2024-12-18:
And i remember the [doc page ](https://dgpu-docs.intel.com/driver/client/overview.html) before has a guide in installing i386 package for Steam Gaming , But the newer doc page delete the Guide and dont have guide in i386 package installing.

eero-t, 2024-12-18:
> 3.GPU: A750
> ...
> 1. Clean Install Of Ubuntu24.04LTS
> ...
> After the install of OS, I install the computing package because i want to use oneAPI and ipex according to the official page
https://dgpu-docs.intel.com/driver/client/overview.html

@ca1ic0 Ubuntu 24.04 has new enough GPU drivers to support A750.

So you do not need to (and should avoid) installing alternative driver versions from Intel repositories.

While user-space drivers in Intel repository offer the same functionality as ones in Ubuntu repositories, they are NOT interchangeable [1].

If some software requires driver package & lib names available only from Intel repo packages, I think it's better to file a bug that they should provide also versions built against distro package & lib names.

(I filed one for XPUManager: https://github.com/intel/xpumanager/issues/89)

While Intel package & lib names came first, and distros chose to use different ones later, I think packages in Intel package repository should be updated to align with distro conventions, and provide transition packages for people upgrading from the old Intel package & lib names,  to avoid these kind of conflicts.

----

[1] Issues:
* They are compiled for the uAPI of the (out-of-tree) i915 DKMS GPU kernel driver in that same repository, so they may not work with i915 distro kernel
* They can have different package names, and different library file names.   While OpenCL and Level-Zero API frontends should abstract this, some SW packages add direct dependencies to backend libraries files and package names, and break when those change
* They may have different package name, but same i.e. conflicting library names, meaning that package installs fail

ca1ic0, 2024-12-18:
> > 3.GPU: A750
> > ...
> > 
> > 1. Clean Install Of Ubuntu24.04LTS
> >    ...
> >    After the install of OS, I install the computing package because i want to use oneAPI and ipex according to the official page
> >    https://dgpu-docs.intel.com/driver/client/overview.html
> 
> @ca1ic0 Ubuntu 24.04 has new enough GPU drivers to support A750.
> 
> So you do not need to (and should avoid) installing alternative driver versions from Intel repositories.
> 
> While user-space drivers in Intel repository offer the same functionality as ones in Ubuntu repositories, they are NOT interchangeable [1].
> 
> If some software requires driver package & lib names available only from Intel repo packages, I think it's better to file a bug that they should provide also versions built against distro package & lib names.
> 
> (I filed one for XPUManager: [intel/xpumanager#89](https://github.com/intel/xpumanager/issues/89))
> 
> While Intel package & lib names came first, and distros chose to use different ones later, I think packages in Intel package repository should be updated to align with distro conventions, and provide transition packages for people upgrading from the old Intel package & lib names, to avoid these kind of conflicts.
> 
> [1] Issues:
> 
> * They are compiled for the uAPI of the (out-of-tree) i915 DKMS GPU kernel driver in that same repository, so they may not work with i915 distro kernel
> * They can have different package names, and different library file names.   While OpenCL and Level-Zero API frontends should abstract this, some SW packages add direct dependencies to backend libraries files and package names, and break when those change
> * They may have different package name, but same i.e. conflicting library names, meaning that package installs fail

Ok, i Got it . This time i didn't add intel repo in apt and install media package from the ubuntu official repo, steam install works fine. Maybe the document could be improved like added some notice or more information?

ca1ic0, 2024-12-18:
later i will try steam gaming

JablonskiMateusz, 2024-12-18:
> 1. clean install the OS
> 2. install `libze1 intel-level-zero-gpu intel-opencl-icd clinfo`
> 3. install the Steam
> 4. launch Steam
>    behave is video [...]

I understand the scenario here, but you have some issues with packages needed for steam, and if it is not visible without compute runtime, it should also manifest in reverse way, i.e. if you have successfully installed stream and then you install compute-runtime packages, there should be some conflict on packages, I wonder which packages are conflicting

eero-t, 2024-12-18:
> This time i didn't add intel repo in apt and install media package from the ubuntu official repo, steam install works fine. Maybe the document could be improved like added some notice or more information?

The instructions at start of that doc did state that you should use distro packages (as your HW is supported):
https://dgpu-docs.intel.com/driver/client/overview.html#selecting-the-right-operating-system-version

> I understand the scenario here, but you have some issues with packages needed for steam, and if it is not visible without compute runtime, it should also manifest in reverse way, i.e. if you have [...]

@JablonskiMateusz here's somewhat similar discussion in XPUM ticket: https://github.com/intel/xpumanager/issues/74#issuecomment-2547237985

Because there's no better Intel GPU monitoring solution, people install `xpu-smi`, but as it's not as actively maintained as `compute-runtime` project, it has deps to conflicting packages.

=>  IMHO `zello_sysman` binary could be provided by default compute-runtime L0 install, so that people have at least  some working GPU metrics tool along with the driver package.

PS. XPUM is a bit awkward for distros.  It install Python modules with pip, and last I checked, depends on versions that would need to be come from multiple Ubuntu versions, partly because Ubuntu includes completely broken ones: https://bugs.launchpad.net/ubuntu/+source/grpc/+bug/1971114

JablonskiMateusz, 2024-12-19:
The XPUM issue sounds like a separate problem.

Let me summarize what was concluded here. Please correct me if I missed anything.

1. @ca1ic0 states that there is conflict on packages when compute-runtime is installed and then steam is installed/used 
2. @eero-t states there is package conflict in XPUM 
3. @eero-t states that `zello_sysman` binary could be somehow delivered to provide very basic GPU metrics tool

Regarding 1. 
1. issue is not seen if compute-runtime is not installed
2. @ca1ic0  please provide dpkg installation error log to show which packages are conflicting
3. @ca1ic0  please try to install compute-runtime packages after stream is working fine - if there is still an issue, please share installation log

Regarding 2.
1. The conflict is on igsc package which is not a part of compute-runtime releases
2. @eero-t  please create corresponding issue in https://github.com/intel/igsc 

Regarding 3.
1. The idea sounds reasonable
2. @eero-t please create a separate issue(feature request) in our GitHub so we can discuss it internally

eero-t, 2024-12-19:
> @eero-t please create a separate issue(feature request) in our GitHub so we can discuss it internally

Done: https://github.com/intel/compute-runtime/issues/787

ca1ic0, 2024-12-23:
> The XPUM issue sounds like a separate problem.
> 
> 让我总结一下这里的结论。如果我遗漏了什么，请纠正我。
> 
> 1. @ca1ic0 states that there is conflict on packages when compute-runtime is installed and then steam is installed/used
> 2. @eero-t states there is package conflict in XPUM
> 3. @eero-t states that `zello_sysman` binary could be somehow delivered to provide very basic GPU metrics tool
> 
> Regarding 1.
> 
> 1. issue is not seen if compute-runtime is not installed
> 2. @ca1ic0  please provide dpkg installation error log to show which packages are conflicting
> 3. @ca1ic0  please try to install compute-runtime packages after stream is working fine - if there is still an issue, please share installation log
> 
> Regarding 2.
> 
> 1. The conflict is on igsc package which is not a part of compute-runtime releases
> 2. @eero-t  please create corresponding issue in https://github.com/intel/igsc
> 
> Regarding 3.
> 
> 1. The idea sounds reasonable
> 2. @eero-t please create a separate issue(feature request) in our GitHub so we can discuss it internally

OK， i will provide a more detailed log in 2025.1.1 vacation.
 And thanks for your support!

pwilma, 2025-10-08:
Hi @ca1ic0 ,
We’d like to know if this issue is still affecting you. If so, please provide an update or any additional information. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated!

closed 2025-11-19 by JablonskiMateusz