intel/compute-runtime #795: What is relationship between driver and compute runtime?

alanzhai219, 2025-01-17:
Hi,
I just read the guide https://dgpu-docs.intel.com/driver/client/overview.html.
Also, there is an install guide on the compute-runtime release page https://github.com/intel/compute-runtime/releases.

What is the relationship between them? Thanks.

alanzhai219, 2025-01-21:
ping

PiotrRozenfeld, 2025-01-29:
Installation instructions provided with compute-runtime releases are focused on getting Compute (Level Zero & OpenCL) APIs functional. 
dgpu-docs.intel.com instructions are wider in scope, and include the Compute packages as well as Media and 3D drivers.

alanzhai219, 2025-01-29:
Could you share the map relationship of deb name and apt install name？

eero-t, 2025-02-13:
Note that (Debian, Ubuntu, Fedora...) distributions package names for compute-runtime may differ from package names used in this project, and packages Intel repos. At least they've historically differed.

When distros started packaging Intel `compute-runtime`, they chose different package names than what `compute-runtime` project itself already used.  I would assume `compute-runtime` and Intel repos will eventually adopt the new names used by corresponding distros, but for a while there can still be dependency issues, e.g. until `xpu-smi` package is updated, or driver packages use `Provides` field for their name(s).

closed 2025-02-14 by alanzhai219