intel/compute-runtime #796: Question-Bug?: Support for OpenGl/OpenCl Interop on B580

catcatmu, 2025-01-17:
Does this driver support the use of OpenGl/OpenCl interoperability for battlemage cards. The card reports the cl_khr_gl_sharing extension but fails all  conformance test. Da Vinci Resolve appears to fail with gl/cl sharing errors as well.

### Versions
Package|Versions|
|--|--|
intel-compute-runtime|24.52.32224.5-1
kernel| 6.12.9-arch1-1|
mesa| 1:24.3.3-2
intel-gmmlib|22.5.5-1

## Ocl Info:
```
Number of platforms                               1
  Platform Name                                   Intel(R) OpenCL Graphics
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 3.0 
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_byte_addressable_store cl_khr_device_uuid cl_khr_fp16 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_intel_command_queue_families cl_intel_subgroups cl_intel_required_subgroup_size cl_intel_subgroups_short cl_khr_spir cl_intel_accelerator cl_intel_driver_diagnostics cl_khr_priority_hints cl_khr_throttle_hints cl_khr_create_command_queue cl_intel_subgroups_char cl_intel_subgroups_long cl_khr_il_program cl_intel_mem_force_host_memory cl_khr_subgroup_extended_types cl_khr_subgroup_non_uniform_vote cl_khr_subgroup_ballot cl_khr_subgroup_non_uniform_arithmetic cl_khr_subgroup_shuffle cl_khr_subgroup_shuffle_relative cl_khr_subgroup_clustered_reduce cl_intel_device_attribute_query cl_khr_extended_bit_ops cl_khr_suggested_local_work_size cl_intel_split_work_group_barrier cl_khr_fp64 cl_khr_subgroups cl_intel_spirv_subgroups cl_khr_spirv_linkonce_odr cl_khr_spirv_no_integer_wrap_decoration cl_intel_unified_shared_memory cl_khr_mipmap_image cl_khr_mipmap_image_writes cl_ext_float_atomics cl_khr_external_memory cl_intel_planar_yuv cl_intel_packed_yuv cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_3d_image_writes cl_intel_bfloat16_conversions cl_intel_create_buffer_with_properties cl_intel_subgroup_local_block_io cl_intel_subgroup_matrix_multiply_accumulate cl_intel_subgroup_matrix_multiply_accumulate_tf32 cl_khr_subgroup_named_barrier cl_intel_subgroup_extended_block_read cl_intel_subgroup_2d_block_io cl_intel_subgroup_buffer_prefetch cl_khr_integer_dot_product cl_khr_gl_sharing cl_khr_gl_depth_images cl_khr_gl_event cl_khr_gl_msaa_sharing cl_intel_va_api_media_sharing cl_intel_sharing_format_query cl_khr_pci_bus_info
......Skipped Lines
Number of devices                                 1
  Device Name                                     Intel(R) Graphics [0xe20b]
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 3.0 NEO 
  Device UUID                                     86800be2-0000-0000-0b00-000000000000
  Driver UUID                                     32342e35-322e-3332-3232-340000000000
  Valid Device LUID                               No
  Device LUID                                     2061-6c78fe7f0000
  Device Node Mask                                0
  Device Numeric Version                          0xc00000 (3.0.0)
  Driver Version                                  24.52.32224
  Device OpenCL C Version                         OpenCL C 1.2 
...... Skipped lines
 Device Extensions                               cl_khr_byte_addressable_store cl_khr_device_uuid cl_khr_fp16 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_intel_command_queue_families cl_intel_subgroups cl_intel_required_subgroup_size cl_intel_subgroups_short cl_khr_spir cl_intel_accelerator cl_intel_driver_diagnostics cl_khr_priority_hints cl_khr_throttle_hints cl_khr_create_command_queue cl_intel_subgroups_char cl_intel_subgroups_long cl_khr_il_program cl_intel_mem_force_host_memory cl_khr_subgroup_extended_types cl_khr_subgroup_non_uniform_vote cl_khr_subgroup_ballot cl_khr_subgroup_non_uniform_arithmetic cl_khr_subgroup_shuffle cl_khr_subgroup_shuffle_relative cl_khr_subgroup_clustered_reduce cl_intel_device_attribute_query cl_khr_extended_bit_ops cl_khr_suggested_local_work_size cl_intel_split_work_group_barrier cl_khr_fp64 cl_khr_subgroups cl_intel_spirv_subgroups cl_khr_spirv_linkonce_odr cl_khr_spirv_no_integer_wrap_decoration cl_intel_unified_shared_memory cl_khr_mipmap_image cl_khr_mipmap_image_writes cl_ext_float_atomics cl_khr_external_memory cl_intel_planar_yuv cl_intel_packed_yuv cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_3d_image_writes cl_intel_bfloat16_conversions cl_intel_create_buffer_with_properties cl_intel_subgroup_local_block_io cl_intel_subgroup_matrix_multiply_accumulate cl_intel_subgroup_matrix_multiply_accumulate_tf32 cl_khr_subgroup_named_barrier cl_intel_subgroup_extended_block_read cl_intel_subgroup_2d_block_io cl_intel_subgroup_buffer_prefetch cl_khr_integer_dot_product cl_khr_gl_sharing cl_khr_gl_depth_images cl_khr_gl_event cl_khr_gl_msaa_sharing cl_intel_va_api_media_sharing cl_intel_sharing_format_query cl_khr_pci_bus_info 

```
## Logs
### OpenGl/Cl conformance test suite
```
redirect_images_read_2D...
Testing image read for GL format GL_TEXTURE_2D : GL_RGBA8 : GL_RGBA : GL_UNSIGNED_BYTE
- Read [  10 x   10 x    1 x    1] : GL Texture : GL_RGBA : GL_RGBA8 : GL_UNSIGNED_BYTE => CL Image : CL_RGBA : CL_UNORM_INT8 
ERROR: Data sample 0,0,0,0 did not validate!
	Expected: 0.671875 0.183594 0.457031 0.750000
	        : 0x1.58p-1 0x1.78p-3 0x1.d4p-2 0x1.8p-1
	  Actual: 0.000000 0.000000 0.000000 0.000000
	        : 0x0p+0 0x0p+0 0x0p+0 0x0p+0
ERROR: Image read test failed for GL_TEXTURE_2D : GL_RGBA8 : GL_RGBA : GL_UNSIGNED_BYTE

Testing image read for GL format GL_TEXTURE_2D : GL_DEPTH_COMPONENT16 : GL_DEPTH_COMPONENT : GL_UNSIGNED_SHORT
[1]    16370 segmentation fault (core dumped)  ./test_gl

```
[gl_tests.txt](https://github.com/user-attachments/files/18459614/gl_tests.txt)

### Resolve Logs:
```
0x77a14edfb000 | DVIP                 | ERROR | 2025-01-17 15:33:51,277 | Failed to register OpenGL object for OpenCL interop: CL_INVALID_GL_OBJECT.
0x77a14edfb000 | UI.GLTexPool         | ERROR | 2025-01-17 15:33:51,277 | Failed to register OpenGL texture to compute API
```

smunaut, 2025-02-06:
It's probably a new tiling format ...
cl/gl sharing isn't really "officially" supported, it's been contributed by community so if you want it fixed, you'll have to write the patch.

smunaut, 2025-02-07:
You could try those mesa MR : 

https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/30670
https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/32120

They might actually make the texture exported with proper modifier that the compute-runtime can accept. 

Please report back if you try them.

catcatmu, 2025-02-07:
I can't currently test it. So I will close this issue for now but update it after if the above fixes the issues.

closed 2025-02-07 by catcatmu

catcatmu, 2025-03-03:
https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/30670  fixes the issue for resolve.