intel/compute-runtime #797: iGPU is not detected

rgov, 2025-01-23:
I'm attempting to run an OpenCL workload in an Ubuntu 22.04 container on a [Intel Celeron J4125](https://www.intel.com/content/www/us/en/products/sku/197305/intel-celeron-processor-j4125-4m-cache-up-to-2-70-ghz/specifications.html) (Gemini Lake) CPU with integrated UHD Graphics 600 GPU.

I am starting the container with `--device=/dev/dri` to mount the DRI device directory into the container, and the `intel-opencl-icd` package is installed. I confirmed that the `i915` driver is loaded.

```
# ls -l /dev/dri
crw------- 1 root root 226,   0 Jan 23 14:56 card0
crw-rw---- 1 root  937 226, 128 Jan 23 14:56 renderD128

```

When I run `sycl-ls` or `clinfo`, only the CPU is discovered:

```
# sycl-ls
[opencl:cpu][opencl:0] Intel(R) OpenCL, Intel(R) Celeron(R) J4125 CPU @ 2.00GHz OpenCL 3.0 (Build 0) [2024.18.7.0.11_160000]
```

The GPU driver is loaded:

```# lspci -nnk

```

rgov, 2025-01-23:
I can get it to work in a new container like this. So maybe it's `intelanalytics/ipex-llm-inference-cpp-xpu` that's broken somehow.

```
$ sudo docker run --rm -it --device=/dev/dri ubuntu:22.04 bash
# apt-get update && apt-get install -y clinfo intel-opencl-icd
Setting up libigdgmm12:amd64 (22.1.2+ds1-1) ...
Setting up libmd0:amd64 (1.0.4-1build1) ...
Setting up ocl-icd-libopencl1:amd64 (2.2.14-3) ...
Setting up libbsd0:amd64 (0.11.5-1) ...
Setting up libicu70:amd64 (70.1-2) ...
Setting up libedit2:amd64 (3.1-20210910-1build1) ...
Setting up clinfo (3.0.21.02.21-1) ...
Setting up libxml2:amd64 (2.9.13+dfsg-1ubuntu0.4) ...
Setting up libllvm12:amd64 (1:12.0.1-19ubuntu3) ...
Setting up libllvmspirvlib12:amd64 (12.0.0-3) ...
Setting up libclang-cpp12 (1:12.0.1-19ubuntu3) ...
Setting up libopencl-clang12:amd64 (12.0.0-3) ...
Setting up libigc1:amd64 (1.0.10840-1) ...
Setting up libigdfcl1:amd64 (1.0.10840-1) ...
Setting up intel-opencl-icd (22.14.22890-1) ...

# clinfo
...
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  Success (1)
    Platform Name                                 Intel(R) OpenCL HD Graphics
    Device Name                                   Intel(R) UHD Graphics 600 [0x3185]
```

rgov, 2025-01-23:
If I roll back to `intel-opencl-icd=23.13.26032.26-627~22.04`, then I can get `clinfo` to identify the GPU platform:

```
# clinfo
  Platform Name                                   Intel(R) OpenCL HD Graphics
Number of devices                                 1
  Device Name                                     Intel(R) UHD Graphics 600
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 3.0 NEO 
...

# sycl-ls
[opencl:cpu][opencl:0] Intel(R) OpenCL, Intel(R) Celeron(R) J4125 CPU @ 2.00GHz OpenCL 3.0 (Build 0) [2024.18.7.0.11_160000]
[opencl:gpu][opencl:1] Intel(R) OpenCL HD Graphics, Intel(R) UHD Graphics 600 OpenCL 3.0 NEO  [23.13.26032.26]
```

I observe that `/usr/local/lib/libigc.so.1` is installed by `intel-igc-core=1.0.17791.9` which shadows the library installed by the `libigc1` package, so to test a downgrade I have to force it to load the correct library:

```
# apt-get install --allow-downgrades -y intel-opencl-icd=23.13.26032.26-627~22.04
# LD_PRELOAD=/lib/x86_64-linux-gnu/libigc.so.1 clinfo
```

sr99622, 2025-01-31:
This is typical garbage I have come to expect from Intel. As they circle the drain, the quality steadily declines. I just did a fresh install Ubuntu and not really surprised but no NPU here. Same machine I have used before and one of the earlier versions worked fine but not this one. By the way the previous version had some bug in GPU driver that caused crashes. Oh well I guess they really don't care too much. Hope you're not holding their stock ;)

JablonskiMateusz, 2025-02-04:
Hi @rgov  
How do you install the 24.39.31294.12 package? 
Isn't that related to the [this](https://github.com/intel/compute-runtime/blob/master/LEGACY_PLATFORMS.md)?

JablonskiMateusz, 2025-02-04:
Also, packages from release 24.39 has known issue described here: https://github.com/intel/compute-runtime/issues/770

rgov, 2025-02-04:
I was just pulling `intelanalytics/ipex-llm-inference-cpp-xpu` which has the 24.39.31294.12 version installed.

The wiki page indeed suggests that my system is not supported -- but how would a user reasonably find this? Even I, who was motivated enough to spend at least **several hours** digging around these repositories, fiddling with different package combinations, searching the web, etc., did not come across this wiki page. It is very unlikely anyone else would.

The newer packages should detect an unsupported legacy iGPU and write something to dmesg or wherever so that the incompatibility can be diagnosed.

JablonskiMateusz, 2025-02-12:
@rgov could you retry on latest `intelanalytics/ipex-llm-inference-cpp-xpu` docker iamge? 

The packages should be updated now https://github.com/intel/ipex-llm/pull/12802

rgov, 2025-02-12:
> [@rgov](https://github.com/rgov) could you retry on latest `intelanalytics/ipex-llm-inference-cpp-xpu` docker iamge?

Yes, I can give it a try. Should I expect to see any acceleration, though, if Gemini Lake does not support Level Zero?

rgov, 2025-02-13:
I gave it a try, but this issue still exists: https://github.com/intel/ipex-llm/issues/12739

JablonskiMateusz, 2025-04-24:
closing as device is detected correctly

closed 2025-04-24 by JablonskiMateusz