intel/compute-runtime #803: Crashes when application tries to read memory info

qnixsynapse, 2025-02-04:
This is the error after the crash:

```
Abort was called at 1060 line in file:
/usr/src/debug/intel-compute-runtime/compute-runtime-24.52.32224.5/shared/source/os_interface/linux/drm_neo.cpp
Aborted (core dumped)
```

When analyzing the stack:
```
#3  0x000077db56033582 in __GI_abort () at abort.c:73
        act = {__sigaction_handler = {sa_handler = 0x424, sa_sigaction = 0x424}, sa_mask = {__val = {131783850941600, 131783925645728, 0, 108160842766240, 108160842763584, 108160842766248,
              108160842731288, 140726044120800, 131783847489969, 140726044120952, 0, 0, 14261692118253170688, 131783926445504, 131783926435824, 140726044120800}}, sa_flags = 1443421989,
          sa_restorer = 0x8}
#4  0x000077db51418081 in NEO::abortExecution () at /usr/src/debug/intel-compute-runtime/compute-runtime-24.52.32224.5/shared/source/helpers/abort.cpp:14
No locals.
#5  NEO::abortUnrecoverable (line=<optimized out>, file=<optimized out>) at /usr/src/debug/intel-compute-runtime/compute-runtime-24.52.32224.5/shared/source/helpers/debug_helpers.cpp:27
No locals.
#6  0x000077db518c629a in NEO::Drm::queryMemoryInfo (this=<optimized out>)
    at /usr/src/debug/intel-compute-runtime/compute-runtime-24.52.32224.5/shared/source/os_interface/linux/drm_neo.cpp:1060
No locals.
#7  NEO::Drm::queryMemoryInfo (this=<optimized out>) at /usr/src/debug/intel-compute-runtime/compute-runtime-24.52.32224.5/shared/source/os_interface/linux/drm_neo.cpp:1059
No locals.
#8  0x000077db516ab36f in L0::LinuxMemoryImp::getState (this=0x625f28b2c860, pState=0x7ffd55ded9d0)
    at /usr/src/debug/intel-compute-runtime/compute-runtime-24.52.32224.5/level_zero/tools/source/sysman/memory/linux/os_memory_imp_dg1.cpp:46
        deviceRegions = std::vector of length 0, capacity 0
        memoryInfo = <optimized out>
#9  0x000077db54d4f735 in urDeviceGetInfo () from /opt/intel/oneapi/compiler/2025.0/lib/libur_adapter_level_zero.so.0
No symbol table info available.
#10 0x000077db558ee81d in ur_loader::urDeviceGetInfo(ur_device_handle_t_*, ur_device_info_t, unsigned long, void*, unsigned long*) ()
   from /opt/intel/oneapi/compiler/2025.0/lib/libur_loader.so.0
No symbol table info available.
#11 0x000077db558fe09d in urDeviceGetInfo () from /opt/intel/oneapi/compiler/2025.0/lib/libur_loader.so.0
No symbol table info available.
#12 0x000077db567ed9eb in sycl::_V1::detail::device_impl::has(sycl::_V1::aspect) const () from /opt/intel/oneapi/compiler/2025.0/lib/libsycl.so.8
No symbol table info available.
#13 0x0000625ee8d0738c in printDeviceInfo(sycl::_V1::device const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
No symbol table info available.
#14 0x0000625ee8d05097 in main ()
```

qnixsynapse, 2025-02-04:
@saik-intel Running `sycl-ls --verbose` triggers it on my system.

This env need to be set in some cases: `export ZES_ENABLE_SYSMAN=1`
It started happening all of a sudden and I am unable to figure out why.

JablonskiMateusz, 2025-02-04:
@qnixsynapse  could you share more details about the device you are using?

qnixsynapse, 2025-02-04:
@JablonskiMateusz GPU is Intel Arc A750 LE 8GB VRAM, CPU is Intel i3 Alder Lake.

qnixsynapse, 2025-02-05:
Fixed by recompiling from source with `-DNEO_ENABLE_I915_PRELIM_DETECTION=1`.
Looks like a downstream issue to me. Arch Linux Packaging do not use this flag.
Will close it once I get a comment from a maintainer about this flag.

JablonskiMateusz, 2025-02-06:
i915 prelim detection is non-default feature and our driver should work fine without this setting. However, based on this issue it looks like something went wrong on our side. @saik-intel  we need to fix this on our side, Arch Linux packaging does nothing wrong. 
@qnixsynapse  please keep this issue open until we provide proper fix for that

qnixsynapse, 2025-02-06:
@JablonskiMateusz Okay but it is still weird that recompiling with the flag fixed the issue, albeit with lower performance.

JablonskiMateusz, 2025-02-07:
> [@JablonskiMateusz](https://github.com/JablonskiMateusz) Okay but it is still weird that recompiling with the flag fixed the issue, albeit with little lower performance.

"weird" part is the logic in our [cmake file](https://github.com/intel/compute-runtime/blob/releases/24.52/level_zero/tools/source/sysman/memory/linux/CMakeLists.txt#L13)  where different cpp file is included in case of i915 prelim support and another one in case of no i915 prelim support.

i915 prelim support enabling should just extend the support

ikubetoomuzik, 2025-02-07:
I don't know if you need more information at this point but I am experiencing the same issue trying to follow [this guide from intel on using Ollama with Intel GPUs](https://github.com/intel/ipex-llm/blob/main/docs/mddocs/Quickstart/ollama_quickstart.md).

I am using the same kernel, distro, and Basekit version as OP, however I am using an Intel ARC A770.
Intel Compute Runtime: 24.52.32224.5-1
Linux: 6.13.1
OS: Arch Linux

I see the following message in the `ollama serve` logs:

```
="0 B" memory.required.full="3.3 GiB" memory.required.partial="0 B" memory.required.kv="896.0 MiB" memory.required.allocations="[3.3 GiB]" memory.weights.total="2.4 GiB" memory.weights.repeating="2.1 GiB" memory.weights.nonrepeating="308.2 MiB" memory.graph.full="424.0 MiB" memory.graph.partial="570.7 MiB"
time=2025-02-07T10:54:52.965-05:00 level=INFO source=server.go:380 msg="starting llama server" cmd="/home/<user>/miniforge3/envs/llm-cpp/lib/python3.11/site-packages/bigdl/cpp/libs/ollama runner --model /home/<user>/.ollama/models/blobs/sha256-dde5aa3fc5ffc17176b5e8bdc82f587b24b2678c6c66101bf7da77af9f7ccdff --ctx-size 8192 --batch-size 512 --n-gpu-layers 999 --threads 4 --no-mmap --parallel 4 --port 38585"
time=2025-02-07T10:54:52.965-05:00 level=INFO source=sched.go:449 msg="loaded runners" count=1
time=2025-02-07T10:54:52.965-05:00 level=INFO source=server.go:559 msg="waiting for llama runner to start responding"
time=2025-02-07T10:54:52.965-05:00 level=INFO source=server.go:593 msg="waiting for server to become available" status="llm server error"
ggml_sycl_init: GGML_SYCL_FORCE_MMQ:   no
ggml_sycl_init: SYCL_USE_XMX: yes
ggml_sycl_init: found 1 SYCL devices:
time=2025-02-07T10:54:53.034-05:00 level=INFO source=runner.go:967 msg="starting go runner"
time=2025-02-07T10:54:53.034-05:00 level=INFO source=runner.go:968 msg=system info="CPU : SSE3 = 1 | SSSE3 = 1 | AVX = 1 | AVX_VNNI = 1 | AVX2 = 1 | F16C = 1 | FMA = 1 | LLAMAFILE = 1 | OPENMP = 1 | AARCH64_REPACK = 1 | CPU : SSE3 = 1 | SSSE3 = 1 | AVX = 1 | AVX_VNNI = 1 | AVX2 = 1 | F16C = 1 | FMA = 1 | LLAMAFILE = 1 | OPENMP = 1 | AARCH64_REPACK = 1 | cgo(gcc)" threads=4
Abort was called at 1060 line in file:
/usr/src/debug/intel-compute-runtime/compute-runtime-24.52.32224.5/shared/source/os_interface/linux/drm_neo.cpp
SIGABRT: abort
PC=0x753af86a5624 m=7 sigcode=18446744073709551610
signal arrived during cgo execution

goroutine 13 gp=0xc000604700 m=7 mp=0xc000580008 [syscall]:
runtime.cgocall(0x558a218584e0, 0xc000099b68)
        runtime/cgocall.go:167 +0x4b fp=0xc000099b40 sp=0xc000099b08 pc=0x558a20cb752b
ollama/llama/llamafile._Cfunc_llama_load_model_from_file(0x753a78000de0, {0x0, 0x3e7, 0x1, 0x0, 0x0, 0x0, 0x558a21857ef0, 0xc00059c2a0, 0x0, ...})
        _cgo_gotypes.go:702 +0x50 fp=0xc000099b68 sp=0xc000099b40 pc=0x558a2107a1f0
ollama/llama/llamafile.LoadModelFromFile.func1({0x7ffc0d52b6dc?, 0x0?}, {0x0, 0x3e7, 0x1, 0x0, 0x0, 0x0, 0x558a21857ef0, 0xc00059c2a0, ...})
        ollama/llama/llamafile/llama.go:247 +0x127 fp=0xc000099c68 sp=0xc000099b68 pc=0x558a2107d627
ollama/llama/llamafile.LoadModelFromFile({0x7ffc0d52b6dc, 0x69}, {0x3e7, 0x0, 0x0, 0x0, {0x0, 0x0, 0x0}, 0xc00047b970, ...})
        ollama/llama/llamafile/llama.go:247 +0x2d6 fp=0xc000099db8 sp=0xc000099c68 pc=0x558a2107d316
ollama/llama/runner.(*Server).loadModel(0xc0001a75f0, {0x3e7, 0x0, 0x0, 0x0, {0x0, 0x0, 0x0}, 0xc00047b970, 0x0}, ...)
        ollama/llama/runner/runner.go:859 +0xc5 fp=0xc000099f10 sp=0xc000099db8 pc=0x558a2108a4a5
ollama/llama/runner.Execute.gowrap1()
        ollama/llama/runner/runner.go:1001 +0xda fp=0xc000099fe0 sp=0xc000099f10 pc=0x558a2108c03a
runtime.goexit({})
        runtime/asm_amd64.s:1700 +0x1 fp=0xc000099fe8 sp=0xc000099fe0 pc=0x558a20cc6001
created by ollama/llama/runner.Execute in goroutine 1
        ollama/llama/runner/runner.go:1001 +0xd0d

goroutine 1 gp=0xc0000061c0 m=nil [runnable]:
log.formatHeader(0xc000556570, {0xc1e1a9134214285f?, 0x1e90248?, 0x558a2259eb60?}, {0x0, 0x0}, 0x0, {0x0, 0x0}, 0x0)
        log/log.go:114 +0x159f fp=0xc0004af748 sp=0xc0004af740 pc=0x558a20ec5f3f
log.(*Logger).output(0xc0000c62d0, 0x0, 0x2, 0xc0004afc60)
        log/log.go:237 +0x330 fp=0xc0004af908 sp=0xc0004af748 pc=0x558a20ec6470
log.Println(...)
        log/log.go:405
ollama/llama/runner.Execute({0xc000036130?, 0x0?, 0x0?})
        ollama/llama/runner/runner.go:1026 +0x118f fp=0xc0004afca8 sp=0xc0004af908 pc=0x558a2108bbef
ollama/cmd.NewCLI.func2(0xc0001cb200?, {0x558a2185cdfd?, 0x4?, 0x558a2185ce01?})
        ollama/cmd/cmd.go:1430 +0x45 fp=0xc0004afcd0 sp=0xc0004afca8 pc=0x558a218575c5
github.com/spf13/cobra.(*Command).execute(0xc0004c0908, {0xc00051cc30, 0xf, 0xf})
        github.com/spf13/cobra@v1.8.1/command.go:985 +0xaaa fp=0xc0004afe58 sp=0xc0004afcd0 pc=0x558a20e4b3ca
github.com/spf13/cobra.(*Command).ExecuteC(0xc000546f08)
        github.com/spf13/cobra@v1.8.1/command.go:1117 +0x3ff fp=0xc0004aff30 sp=0xc0004afe58 pc=0x558a20e4bc9f
github.com/spf13/cobra.(*Command).Execute(...)
        github.com/spf13/cobra@v1.8.1/command.go:1041
github.com/spf13/cobra.(*Command).ExecuteContext(...)
        github.com/spf13/cobra@v1.8.1/command.go:1034
main.main()
        ollama/main.go:12 +0x4d fp=0xc0004aff50 sp=0xc0004aff30 pc=0x558a21857c2d
runtime.main()
        runtime/proc.go:272 +0x29d fp=0xc0004affe0 sp=0xc0004aff50 pc=0x558a20c88f3d
runtime.goexit({})
        runtime/asm_amd64.s:1700 +0x1 fp=0xc0004affe8 sp=0xc0004affe0 pc=0x558a20cc6001

[... 159 lines cut ...]
[... cut ...]
ollama/llama/runner.(*Server).run(0xc0001a75f0, {0x558a21e08aa0, 0xc00052dd10})
        ollama/llama/runner/runner.go:315 +0x47 fp=0xc00050d7b8 sp=0xc00050d6e0 pc=0x558a21086b47
ollama/llama/runner.Execute.gowrap2()
        ollama/llama/runner/runner.go:1006 +0x28 fp=0xc00050d7e0 sp=0xc00050d7b8 pc=0x558a2108bf28
runtime.goexit({})
        runtime/asm_amd64.s:1700 +0x1 fp=0xc00050d7e8 sp=0xc00050d7e0 pc=0x558a20cc6001
created by ollama/llama/runner.Execute in goroutine 1
        ollama/llama/runner/runner.go:1006 +0xde5
```

JablonskiMateusz, 2025-02-12:
Unification fix is merged: https://github.com/intel/compute-runtime/commit/e91480c673cb573552dd63b0153cabb988505c72 
@qnixsynapse  could you confirm that it resolves your isssue (when not setting i915 prelim support at cmake level)?

JablonskiMateusz, 2025-02-12:
@qnixsynapse  
> Yes, it seems to fix the issue.

Great to hear that!

> BTW, I also noticed that some Windows stuffs are also getting built on Linux. Not sure if that's intentional or not but anyways, thanks.

It could be related to WSL (WDDM on Linux) support

qnixsynapse, 2025-02-12:
Ah. Okay. Closing this.

qnixsynapse, 2025-04-18:
It has been more than 2 months. Why the the fix is not in stable yet? Is this not a P0 priority fix for you guys?

closed 2025-04-18 by qnixsynapse