intel/compute-runtime #805: Loading of FCL library has failed! Filename: libigdfcl.so.2

bl4ckb0ne, 2025-02-05:
Hi,

I'm packaging the intel opencl icd `24.52.32224.5` for alpine linux, and I encounter the following error repeating over and over.

```
ninja: job failed: cd /builds/alpine/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels && /usr/bin/cmake -E env LD_LIBRARY_PATH=:/builds/alpine/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/build/bin /builds/alpine/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/build/bin/ocloc-24.52.1 -q -file /builds/alpine/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels/copy_image_to_image3d.builtin_kernel -spv_only -device lnl -force_stos_opt -64 -output bindful_copy_image_to_image3d -out_dir /builds/alpine/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/build/bin/built_ins/x64/xe2_hpg_core -options -cl-kernel-arg-info
Error! Loading of FCL library has failed! Filename: libigdfcl.so.2
Error! FCL initialization failure. Error code = -6
Build failed with error code: -6
```

The library is installed in `/usr`

```
$ find /usr/ -name "libigdfcl.so.2"
/usr/lib/libigdfcl.so.2
```

and it is found by CMake

```
-- IGA library name: libiga64.so.2
-- IGA Includes dir: /usr/include/iga
```

The `LD_LIBRARY_PATH` seems wrong, it is missing the system path.

JablonskiMateusz, 2025-02-06:
Hi @bl4ckb0ne  
please share entire cmake log

bl4ckb0ne, 2025-02-06:
```
-- The C compiler identification is Clang 19.1.7
-- The CXX compiler identification is Clang 19.1.7
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- branch dir list: /
-- Host processor: x86_64
-- Target processor: x86_64
-- Neo target processor: x86_64
-- Performing Test COMPILER_SUPPORTS_FLTO_AUTO
-- Performing Test COMPILER_SUPPORTS_FLTO_AUTO - Failed
-- Performing Test COMPILER_SUPPORTS_LLD
-- Performing Test COMPILER_SUPPORTS_LLD - Success
-- Performing Test COMPILER_SUPPORTS_GOLD
-- Performing Test COMPILER_SUPPORTS_GOLD - Success
-- Using linker: default
-- Release build configuration
-- NEO_SKIP_AUB_TESTS: TRUE
-- NEO_SKIP_AUB_TESTS_RUN: TRUE
-- Aub Stream Headers dir: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.52.32224.5/third_party/aub_stream
-- Khronos OpenCL headers dir: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.52.32224.5/third_party/opencl_headers
-- Khronos OpenGL headers dir: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.52.32224.5/third_party/opengl_headers
-- AOT config headers dir: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.52.32224.5/third_party/aot_config_headers
-- Found PkgConfig: /usr/bin/pkg-config (found version "2.3.0")
-- Checking for module 'igc-opencl'
--   Found igc-opencl, version 2.5.0
-- IGC include dirs: /usr/include/igc;/usr/include/igc/cif;/usr/include/igc/ocl_igc_shared/executable_format;/usr/include/igc/ocl_igc_shared/device_enqueue
-- VISA Dir: /usr/include/visa
-- IGA library name: libiga64.so.2
-- IGA Includes dir: /usr/include/iga
-- Checking for module 'igdgmm'
--   Found igdgmm, version 12.6.0
-- GmmLib include dirs: /usr/include/igdgmm;/usr/include/igdgmm/GmmLib;/usr/include/igdgmm/GmmLib/inc;/usr/include/igdgmm/inc;/usr/include/igdgmm/inc/common;/usr/include/igdgmm/util
-- Checking for module 'libigdml'
--   Package 'libigdml' not found
-- Metrics Library dir: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.52.32224.5/third_party/metrics_library
-- Checking for module 'libigdmd'
--   Package 'libigdmd' not found
-- Metrics Discovery dir: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.52.32224.5/third_party/metrics_discovery
-- Checking for module 'libva>=1.0.0'
--   Found libva, version 1.22.0
-- Looking for vaGetLibFunc in va
-- Looking for vaGetLibFunc in va - found
-- Using libva 
-- LibVA include dirs: /usr/include
-- AUB_STREAM_DIR = 
-- Engine node dir: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.52.32224.5/third_party/aub_stream
-- Auto-Enabling GEN12LP support for DG1
-- Auto-Enabling XE_HPG_CORE support for DG2
-- All supported platforms:  BMG LNL PVC MTL DG2 ARL TGLLP DG1 RKL ADLS ADLP ADLN
-- All tested platforms: 
-- Default supported platform: PVC
-- Default tested platform: 
-- All supported core families: GEN12LP;XE_HPG_CORE;XE_HPC_CORE;XE2_HPG_CORE
-- All tested core families: 
-- Default tested family name: 
-- Driver model : drm
-- i915 prelim headers detection: FALSE
-- xe prelim headers detection: FALSE
-- Xe EU Debug support: FALSE
-- drm includes dir: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.52.32224.5/third_party/uapi/upstream/drm
-- i915 includes dir: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.52.32224.5/third_party/uapi/upstream/i915
-- i915 prelim includes dir: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.52.32224.5/third_party/uapi/drm-uapi-helper/master
-- xe includes dir: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.52.32224.5/third_party/uapi/upstream/xe
-- iaf includes dir: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.52.32224.5/third_party/uapi/drm-uapi-helper/master
-- Neo kernels build type dir: release
-- Performing Test COMPILER_SUPPORTS_RETPOLINE
-- Performing Test COMPILER_SUPPORTS_RETPOLINE - Success
-- Performing Test COMPILER_SUPPORTS_SSE42
-- Performing Test COMPILER_SUPPORTS_SSE42 - Success
-- Performing Test COMPILER_SUPPORTS_AVX2
-- Performing Test COMPILER_SUPPORTS_AVX2 - Success
-- Performing Test COMPILER_SUPPORTS_NEON
-- Performing Test COMPILER_SUPPORTS_NEON - Failed
-- Performing Test SUPPORTS_CLFLUSHOPT
-- Performing Test SUPPORTS_CLFLUSHOPT - Success
-- Performing Test SUPPORTS_WAITPKG
-- Performing Test SUPPORTS_WAITPKG - Success
-- [OCLOC] All supported platforms: TGLLP;DG1;RKL;ADLS;ADLP;ADLN;MTL;DG2;ARL;PVC;BMG;LNL
-- ocloc_lib current name: $<TARGET_FILE_NAME:ocloc_lib>
-- ocloc_lib former name: libocloc_legacy1.so
-- L0: Selected platforms for testing are not supported. Skipping tests.
-- Level Zero driver version: 1.6.32224
-- Found LevelZero: /usr/include
-- LevelZero_INCLUDE_DIRS: /usr/include
-- LibGenl headers not available. Building without
-- libudev headers directory: /usr/include
-- igsc Library headers directory: 
-- igsc version: 0.9.5
-- Prebuilt kernels are linked to Level Zero.
-- Configuring done (2.7s)
-- Generating done (1.5s)
-- Build files have been written to: /builds/alpine/aports/testing/intel-opencl-icd/src/compute-runtime-24.52.32224.5/build
```

And these are the CMake arguments

```

	cmake -B build -G Ninja \
		-DCMAKE_BUILD_TYPE=Release \
		-DCMAKE_C_COMPILER=clang \
		-DCMAKE_CXX_COMPILER=clang++ \
		-DCMAKE_INSTALL_PREFIX=/usr \
		-DCMAKE_INSTALL_LIBDIR=lib \
		-DSUPPORT_DG1=ON \
		-DSUPPORT_DG2=ON \
		-DDISABLE_WDDM_LINUX=ON \
		-DNEO_DISABLE_LTO=OFF \
		-DNEO_SKIP_UNIT_TESTS=1 \
		.

```

JablonskiMateusz, 2025-02-10:
@bl4ckb0ne could you get below data?
1. strace log from running the failing command
```
cd /builds/alpine/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels && /usr/bin/cmake -E env LD_LIBRARY_PATH=:/builds/alpine/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/build/bin /builds/alpine/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/build/bin/ocloc-24.52.1 -q -file /builds/alpine/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels/copy_image_to_image3d.builtin_kernel -spv_only -device lnl -force_stos_opt -64 -output bindful_copy_image_to_image3d -out_dir /builds/alpine/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/build/bin/built_ins/x64/xe2_hpg_core -options -cl-kernel-arg-info
```
2. output from `ldd -r /usr/lib/libigdfcl.so.2` 
3. output from `pkg-config --libs igc-opencl`

JablonskiMateusz, 2025-02-10:
Regarding 1. I provided the failing command from the log you mentioned earlier. To get strace log you may run: strace -o strace.log <command> 

However, if the command didn't fail, it means the library was found. That's opposite to what you mentioned at the very beginning.

bl4ckb0ne, 2025-02-10:
Ah right, strace sorry. I think I might've accidentally fix paths when updating the locations. 

Here's the [log.txt](https://github.com/user-attachments/files/18740275/log.txt) for this

```
cd /home/simon/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/build/shared/source/built_ins/kernels && /usr/bin/cmake -E env LD_LIBRARY_PATH=:/home/simon/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/build/bin strace /home/simon/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/build/bin/ocloc-24.52.1 -q -file /home/simon/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels/copy_image_to_image3d.builtin_kernel -spv_only -device lnl -force_stos_opt -64 -output bindful_copy_image_to_image3d -out_dir /home/simon/aports/testing/intel-compute-runtime/src/compute-runtime-24.52.32224.5/build/bin/built_ins/x64/xe2_hpg_core -options -cl-kernel-arg-info
```

JablonskiMateusz, 2025-02-11:
> Ah right, strace sorry. I think I might've accidentally fix paths when updating the locations.
>
> Here's the [log.txt](https://github.com/user-attachments/files/18740275/log.txt) for this
>
> ```
> cd [...]

since this one is working fine, I don't see anything incorrect on this. Please try repro the initial issue on clear env. And if happens, then capture strace log

bl4ckb0ne, 2025-02-11:
I added the line to the package build directly

```
$cd shared/source/built_ins/kernels && cmake -E env LD_LIBRARY_PATH=:/tmp/src/compute-runtime-24.52.32224.5/build/bin strace /tmp/src/compute-runtime-24.52.32224.5/build/bin/ocloc-24.52.1 -q -file /tmp/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels/copy_image_to_image3d.builtin_kernel -spv_only -device lnl -force_stos_opt -64 -output bindful_copy_image_to_image3d -out_dir /tmp/src/compute-runtime-24.52.32224.5/build/bin/built_ins/x64/xe2_hpg_core -options -cl-kernel-arg-info
```

And here's the strace log

```
ninja: job failed: cd /tmp/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels && /usr/bin/cmake -E env LD_LIBRARY_PATH=:/tmp/src/compute-runtime-24.52.32224.5/build/bin /tmp/src/compute-runtime-24.52.32224.5/build/bin/ocloc-24.52.1 -q -file /tmp/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels/copy_kernel_timestamps.builtin_kernel -spv_only -device lnl -force_stos_opt -64 -output bindless_copy_kernel_timestamps -out_dir /tmp/src/compute-runtime-24.52.32224.5/build/bin/built_ins/x64/xe2_hpg_core -internal_options -cl-intel-use-bindless-mode\ -cl-intel-use-bindless-advanced-mode -options -cl-kernel-arg-info
Error! Loading of FCL library has failed! Filename: libigdfcl.so.2
Error! FCL initialization failure. Error code = -6
Build failed with error code: -6
Command was: /tmp/src/compute-runtime-24.52.32224.5/build/bin/ocloc-24.52.1 -q -file /tmp/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels/copy_kernel_timestamps.builtin_kernel -spv_only -device lnl -force_stos_opt -64 -output bindless_copy_kernel_timestamps -out_dir /tmp/src/compute-runtime-24.52.32224.5/build/bin/built_ins/x64/xe2_hpg_core -internal_options "-cl-intel-use-bindless-mode -cl-intel-use-bindless-advanced-mode" -options "-cl-kernel-arg-info"
ninja: job failed: cd /tmp/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels && /usr/bin/cmake -E env LD_LIBRARY_PATH=:/tmp/src/compute-runtime-24.52.32224.5/build/bin /tmp/src/compute-runtime-24.52.32224.5/build/bin/ocloc-24.52.1 -q -file /tmp/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels/fill_buffer.builtin_kernel -spv_only -device lnl -force_stos_opt -64 -output bindless_fill_buffer -out_dir /tmp/src/compute-runtime-24.52.32224.5/build/bin/built_ins/x64/xe2_hpg_core -internal_options -cl-intel-use-bindless-mode\ -cl-intel-use-bindless-advanced-mode -options -cl-kernel-arg-info
Error! Loading of FCL library has failed! Filename: libigdfcl.so.2
Error! FCL initialization failure. Error code = -6
Build failed with error code: -6
Command was: /tmp/src/compute-runtime-24.52.32224.5/build/bin/ocloc-24.52.1 -q -file /tmp/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels/fill_buffer.builtin_kernel -spv_only -device lnl -force_stos_opt -64 -output bindless_fill_buffer -out_dir /tmp/src/compute-runtime-24.52.32224.5/build/bin/built_ins/x64/xe2_hpg_core -internal_options "-cl-intel-use-bindless-mode -cl-intel-use-bindless-advanced-mode" -options "-cl-kernel-arg-info"
ninja: job failed: cd /tmp/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels && /usr/bin/cmake -E env LD_LIBRARY_PATH=:/tmp/src/compute-runtime-24.52.32224.5/build/bin /tmp/src/compute-runtime-24.52.32224.5/build/bin/ocloc-24.52.1 -q -file /tmp/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels/copy_image_to_image3d.builtin_kernel -spv_only -device lnl -force_stos_opt -64 -output bindless_copy_image_to_image3d -out_dir /tmp/src/compute-runtime-24.52.32224.5/build/bin/built_ins/x64/xe2_hpg_core -internal_options -cl-intel-use-bindless-mode\ -cl-intel-use-bindless-advanced-mode -options -cl-kernel-arg-info
Error! Loading of FCL library has failed! Filename: libigdfcl.so.2
Error! FCL initialization failure. Error code = -6
Build failed with error code: -6
Command was: /tmp/src/compute-runtime-24.52.32224.5/build/bin/ocloc-24.52.1 -q -file /tmp/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels/copy_image_to_image3d.builtin_kernel -spv_only -device lnl -force_stos_opt -64 -output bindless_copy_image_to_image3d -out_dir /tmp/src/compute-runtime-24.52.32224.5/build/bin/built_ins/x64/xe2_hpg_core -internal_options "-cl-intel-use-bindless-mode -cl-intel-use-bindless-advanced-mode" -options "-cl-kernel-arg-info"
ninja: job failed: cd /tmp/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels && /usr/bin/cmake -E env LD_LIBRARY_PATH=:/tmp/src/compute-runtime-24.52.32224.5/build/bin /tmp/src/compute-runtime-24.52.32224.5/build/bin/ocloc-24.52.1 -q -file /tmp/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels/fill_image2d.builtin_kernel -spv_only -device lnl -force_stos_opt -64 -output bindless_fill_image2d -out_dir /tmp/src/compute-runtime-24.52.32224.5/build/bin/built_ins/x64/xe2_hpg_core -internal_options -cl-intel-use-bindless-mode\ -cl-intel-use-bindless-advanced-mode -options -cl-kernel-arg-info
Error! Loading of FCL library has failed! Filename: libigdfcl.so.2
Error! FCL initialization failure. Error code = -6
Build failed with error code: -6
Command was: /tmp/src/compute-runtime-24.52.32224.5/build/bin/ocloc-24.52.1 -q -file /tmp/src/compute-runtime-24.52.32224.5/shared/source/built_ins/kernels/fill_image2d.builtin_kernel -spv_only -device lnl -force_stos_opt -64 -output bindless_fill_image2d -out_dir /tmp/src/compute-runtime-24.52.32224.5/build/bin/built_ins/x64/xe2_hpg_core -internal_options "-cl-in [... cut ...] e-runtime-24.52.32224.5/shared/source/built_ins/kernels/copy_image_to_image3d.builtin_kernel -spv_only -device lnl -force_stos_opt -64 -output bindful_copy_image_to_image3d -out_dir /tmp/src/compute-runtime-24.52.32224.5/build/bin/built_ins/x64/xe2_hpg_core -options "-cl-kernel-arg-info"
) = 389
munmap(0x7f3dc85db000, 4096)            = 0
munmap(0x7f3dc85dc000, 8192)            = 0
munmap(0x7f3dc85da000, 4096)            = 0
exit_group(-6)                          = ?
+++ exited with 250 +++
```

JablonskiMateusz, 2025-02-11:
If you are not using prebuilt packages and building everything on your own, what are your commands, what is order of building packages?
Are you building both libigc and libigdfcl before building compute runtime?

JablonskiMateusz, 2025-02-12:
the part of
> error: IGC: 
internal compiler error
Build failed with error code: -11

looks like a separate issue

bl4ckb0ne, 2025-02-12:
I am also packaging the requirements (igsc, level-zero and igc) but I think this is an issue in the package builder.

Regarding the internal compiler error, would you like me to open another issue?

bl4ckb0ne, 2025-02-12:
Seems to be coming from clang, I used GCC instead and it built fine

bl4ckb0ne, 2025-02-12:
I'll close this one too, most likely a clang issue.