intel/compute-runtime #806: libze_intel_gpu.so didn't load when executing sycl-ls

wdkwyf, 2025-02-07:
Hi, I'm using LNL and want to use level-zero iGPU. However, when I install NEO, only opencl environment is good.

KMD:
ubuntu 24.10
> uname -r
> 6.11.0-1006-intel
UMD:
25.04 release
```
ZE_ENABLE_LOADER_DEBUG_TRACE=1 sycl-ls
```
```
ZE_LOADER_DEBUG_TRACE:Using Loader Library Path:
ZE_LOADER_DEBUG_TRACE:Load Library of libze_intel_vpu.so failed with libze_intel_vpu.so: cannot open shared object file: No such file or directory
ZE_LOADER_DEBUG_TRACE:0 Drivers Discovered
ZE_LOADER_DEBUG_TRACE:Using Loader Library Path:
ZE_LOADER_DEBUG_TRACE:Load Library of libze_intel_vpu.so failed with libze_intel_vpu.so: cannot open shared object file: No such file or directory
ZE_LOADER_DEBUG_TRACE:0 Drivers Discovered
[opencl:cpu][opencl:0] Intel(R) OpenCL, Intel(R) Core(TM) Ultra 7 268V OpenCL 3.0 (Build 0) [2025.19.1.0.22_160000]
[opencl:gpu][opencl:1] Intel(R) OpenCL Graphics, Intel(R) Arc(TM) Graphics OpenCL 3.0 NEO  [25.04.32460]
```
libze_intel_gpu.so exist in /lib/x86_64-linux-gnu/, but the level zero only find vpu and didn't load any gpu so file.
If any suggestion, would be appreciate!

nrspruit, 2025-02-07:
Hello @wdkwyf 

So, please check your LD_LIBRARY_PATH, if the loader is only showing the libze_intel_vpu.so, then the libze_intel_gpu.so is missing from the path.

If the gpu driver library was in the path you would have seen:
ZE_LOADER_DEBUG_TRACE:Using Loader Library Path:
ZE_LOADER_DEBUG_TRACE:Load Library of libze_intel_gpu.so

In fact it should always print that unless the loader you are using is missing the lines:
static const char *knownDriverNames[] = {
    MAKE_LIBRARY_NAME("ze_intel_gpu", "1"),
..

In the linux driver discovery.

Can you provide the logs from:

strace sycl-ls

and

SYCL_UR_TRACE=-1 sycl-ls

Thank you for your time!

wdkwyf, 2025-03-26:
> Hello [@wdkwyf](https://github.com/wdkwyf)
>
> So, please check your LD_LIBRARY_PATH, if the loader is only showing the libze_intel_vpu.so, then the libze_intel_gpu.so is missing from the path.
>
> If the [...]

@nrspruit Thanks for your help! With your suggestion, I found the library was broken. Thank you for your professional knowledge, which helped to solve this problem.

closed 2025-03-26 by wdkwyf