intel/compute-runtime #810: Way to query free memory from an openCL backend on discrete Intel GPUs

icfaust, 2025-02-09:
As a developer for the uxlfoundation on the [oneDAL](https://github.com/uxlfoundation/oneDAL) and [scikit-learn-intelex](https://github.com/uxlfoundation/scikit-learn-intelex/), I would like to query the free memory of an Intel GPU when a level zero driver is unavailable, but an openCL is.  This would be especially important for automated testing using Intel Data Center GPU products (Intel Max 1550) when accessing the underlying shared object memory allocations are not possible simply and/or directly.

While the functionality is available in SYCL when the level Zero SysMan is available, there is no equivalent for openCL backends on intel discrete products. However, for other vendors such functionality exists: ```nvmlDeviceGetMemoryInfo``` for NVidia and ```CL_DEVICE_GLOBAL_FREE_MEMORY_AMD``` for Radeon products.

Replicated from this reddit post (could not find the related issues in the compute-runtime closed/open list): https://www.reddit.com/r/OpenCL/comments/uocbbq/comment/i8rcfhs/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

eero-t, 2025-02-12:
@icfaust see https://github.com/intel/compute-runtime/issues/586.

icfaust, 2025-02-13:
@eero-t Thank you for the link, I do appreciate your suggestion [here](https://github.com/intel/compute-runtime/issues/586#issuecomment-1344091067) to use ze_sysman for querying the free memory. We use that successfully already on level-zero enabled discrete intel GPUs, however, in some cases we are trying to verify operation on openCL backends, while the link discusses how to query the total available memory on a device, it does not return the current free memory on a device.