intel/compute-runtime #811: Is the latest  Client GPUs driver working as expected on meteor laker device?

ocean1ee1, 2025-02-12:
I'm trying to run applications which used oneAPI on level zero device.

So from some guides, [the GPU driver](https://dgpu-docs.intel.com/driver/client/overview.html) should be installed first.

And from [hardware table](https://dgpu-docs.intel.com/devices/hardware-table.html), I can find my device is listed in that table

PCI IDs | Name                | Architecture | Codename    | Kernel  | EU number
7D45    | Intel® Graphics | Xe-LPG        | Meteor Lake | 6.8*     | 64/48

The Kernel column represents the version of the upstream Linux kernel that first introduced support for each device. Kernel versions marked with an asterisk (*) require enabling support via the force_probe flag. 

-----------------

My environment is ubuntu 24.04 with a 6.12.13 kernel.

After installed these driver packages, sycl-ls can identify this igpu, however in dmesg, it shows it still use i915 related drivers.

1. Tried to add xe.force_probe=7d45 and i915.force_probe=!7d45 to boot paras to start 6.12.13. I got a black screen and not able to login(even try to switch to command line mode), in this time I can use ctrl + alt + del to reboot.
2. Tried to only add i915.force_probe=!7d45 to boot paras to start 6.12.13, wait a moment, it will let me login with a basic graphic driver. in dmesg, it shows my device is not officially support to use xe.
3. Tried to add xe.force_probe=7d45 and i915.force_probe=!7d45 to boot paras to start 6.8.0, it will let me login, but in dmesg, it shows errors, and I cannot run application that required level zero device.
6.8 kernel dmesg:
[    3.063764] xe 0000:00:02.0: vgaarb: deactivate vga console
[    3.076925] xe 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=io+mem
[    3.080849] xe 0000:00:02.0: [drm] HDCP support not yet implemented
[    3.090831] xe 0000:00:02.0: [drm] Finished loading DMC firmware i915/mtl_dmc.bin (v2.21)
[    3.353625] xe 0000:00:02.0: [drm] Using GuC firmware from i915/mtl_guc_70.bin version 70.20.0
[    3.459489] xe 0000:00:02.0: [drm] Using GuC firmware from i915/mtl_guc_70.bin version 70.20.0
[    3.466550] xe 0000:00:02.0: [drm] Using HuC firmware from i915/mtl_huc_gsc.bin version 8.5.4
[    3.476128] xe 0000:00:02.0: [drm] Using GSC firmware from i915/mtl_gsc_1.bin version 102.0.10.1878
[    3.784019] xe 0000:00:02.0: [drm] Selective fetch area calculation failed in pipe A
[    3.921435] xe 0000:00:02.0: [drm] GT1: found GSC cv1.0
[    5.640247] [drm] Initialized xe 1.1.0 20201103 for 0000:00:02.0 on minor 1
[    5.646419] xe 0000:00:02.0: [drm] Allocated fbdev into stolen
[    5.654215] xe 0000:00:02.0: [drm] fb1: xedrmfb frame buffer device
[   20.638486] xe 0000:00:02.0: [drm] *ERROR* [CRTC:80:pipe A] DSB 0 timed out waiting for idle (current head=0x4c0, head=0x0, tail=0x1080)

Since I'm new to this area, are these normal ? the meteor devices are only supported by i915?
Or is there other things I can try, or other information I need to provide ?

JablonskiMateusz, 2025-02-12:
Hi @ocean1ee1  
Meteorlake platform is supported by upstream i915 KMD
Xe-KMD supports Lunarlake, Battlemage and onwards platforms

ocean1ee1, 2025-02-12:
Thanks. Mateusz

May I know in future, will this change? Is there any plan or rumor to make Xe-KMD to Meteor lake in sometime?

ocean1ee1, 2025-02-13:
Eero, thank you for explanation and time. I will close this