intel/compute-runtime #812: [level0] fabs(-0.0) returns -0.0

linehill, 2025-02-13:
But the result should be positive zero, right?

SPIR-V assembly:
```
; SPIR-V
; Version: 1.2
; Generator: Khronos LLVM/SPIR-V Translator; 14
; Bound: 42
; Schema: 0
               OpCapability Addresses
               OpCapability Linkage
               OpCapability Kernel
               OpCapability Int64
               OpCapability Int8
          %1 = OpExtInstImport "OpenCL.std"
               OpMemoryModel Physical64 OpenCL
               OpEntryPoint Kernel %39 "k"
               OpSource OpenCL_C 300000
               OpName %_str ".str"
               OpName %k "k"
               OpDecorate %_str Constant
               OpDecorate %_str Alignment 1
               OpDecorate %k LinkageAttributes "k" Export
      %ulong = OpTypeInt 64 0
      %uchar = OpTypeInt 8 0
       %uint = OpTypeInt 32 0
   %ulong_27 = OpConstant %ulong 27
  %uchar_114 = OpConstant %uchar 114
  %uchar_101 = OpConstant %uchar 101
  %uchar_115 = OpConstant %uchar 115
   %uchar_61 = OpConstant %uchar 61
   %uchar_37 = OpConstant %uchar 37
  %uchar_102 = OpConstant %uchar 102
   %uchar_44 = OpConstant %uchar 44
   %uchar_32 = OpConstant %uchar 32
   %uchar_97 = OpConstant %uchar 97
   %uchar_95 = OpConstant %uchar 95
  %uchar_117 = OpConstant %uchar 117
  %uchar_105 = OpConstant %uchar 105
  %uchar_110 = OpConstant %uchar 110
  %uchar_116 = OpConstant %uchar 116
   %uchar_40 = OpConstant %uchar 40
   %uchar_41 = OpConstant %uchar 41
   %uchar_48 = OpConstant %uchar 48
  %uchar_120 = OpConstant %uchar 120
   %uchar_10 = OpConstant %uchar 10
    %uchar_0 = OpConstant %uchar 0
%_arr_uchar_ulong_27 = OpTypeArray %uchar %ulong_27
%_ptr_UniformConstant__arr_uchar_ulong_27 = OpTypePointer UniformConstant %_arr_uchar_ulong_27
       %void = OpTypeVoid
         %30 = OpTypeFunction %void
      %float = OpTypeFloat 32
         %27 = OpConstantComposite %_arr_uchar_ulong_27 %uchar_114 %uchar_101 %uchar_115 %uchar_61 %uchar_37 %uchar_102 %uchar_44 %uchar_32 %uchar_97 %uchar_115 %uchar_95 %uchar_117 %uchar_105 %uchar_110 %uchar_116 %uchar_40 %uchar_114 %uchar_101 %uchar_115 %uchar_41 %uchar_61 %uchar_48 %uchar_120 %uchar_37 %uchar_120 %uchar_10 %uchar_0
       %_str = OpVariable %_ptr_UniformConstant__arr_uchar_ulong_27 UniformConstant %27
   %float_n0 = OpConstant %float -0
          %k = OpFunction %void None %30
         %32 = OpLabel
         %35 = OpExtInst %float %1 fabs %float_n0
         %37 = OpBitcast %uint %35
         %38 = OpExtInst %uint %1 printf %_str %35 %37
               OpReturn
               OpFunctionEnd
         %39 = OpFunction %void None %30
         %40 = OpLabel
         %41 = OpFunctionCall %void %k
               OpReturn
               OpFunctionEnd
```

Runtime:
```
Driver version: 17005720
Selected device: 'Intel(R) Iris(R) Xe Graphics'
FAIL: res=-0.000000, as_uint(res)=0x80000000. Expected positive zero.
```

RocketRide9, 2025-04-08:
If it was OpenCL, it would look like you used `-cl-no-signed-zeros` compiler flag: https://registry.khronos.org/OpenCL/specs/3.0-unified/html/OpenCL_API.html#optimization-options. I don't have any experience with level0. Maybe you should look if you used similar compiler option?

linehill, 2025-04-08:
The kernel is built without build options and run through level zero API. AFAICT, building the kernel without build options defaults to `-O2` but that does not enable transformations similar to the `-cl-no-signed-zero`.

pwilma, 2025-10-01:
@linehill this issue was fixed some time ago in IGC:
https://github.com/intel/intel-graphics-compiler/commit/b278ddd68eb4d04fbeffb6d7fc0a441c758e2431
Could you please check if the problem is still reproducible with latest Neo release?
https://github.com/intel/compute-runtime/releases/tag/25.35.35096.9