intel/compute-runtime #815: error: ‘IGFX_XE3_CORE’ was not declared in this scope; did you mean ‘IGFX_GEN3_CORE’?

picarica, 2025-03-11:
hello i have issues compiling this intel-compute-runtime-25.05.32567.18 in gentoo, i am not sure if this is gentoo specific driver, i thought maybe i could ask here also
original forum post i posted [here](https://forums.gentoo.org/viewtopic-t-1173280.html) 
[here is my build.log](https://paste.gentoo.zip/vJLh58In)
[and some info with what options i compiled said package](https://paste.gentoo.zip/Sw0LttM7)
how would i go fix about, do i need to write some patch ?

HoppeMateusz, 2025-03-13:
Hello

are you using correct gmm lib ? [intel/gmmlib@intel-gmmlib-22.6.0](https://github.com/intel/gmmlib/releases/tag/intel-gmmlib-22.6.0)

picarica, 2025-03-13:
i noticed >=media-libs/gmmlib-22.5.2: this in the ebuild  i will try adding 22.6.0 and will update

picarica, 2025-03-13:
yep that was it, i changed verions to 22.7.0, and it compiled fine, thank you for your support i already filed a bug report
https://forums.gentoo.org/viewtopic-p-8856959.html#8856959
https://bugs.gentoo.org/951283

closed 2025-03-13 by picarica