intel/compute-runtime #816: Undefined behavior for deviceInfo reporting in case of unsupported extensions

ProjectPhysX, 2025-03-12:
All [`deviceInfo` members](https://github.com/intel/compute-runtime/blob/master/opencl/source/cl_device/cl_device_info.h) are missing default-initialization with zero.

Moreover, for `deviceInfo.integerDotCapabilities` / `deviceInfo.integerDotAccelerationProperties8Bit` / `deviceInfo.integerDotAccelerationProperties4x8BitPacked`, the [else-branch is missing here](https://github.com/intel/compute-runtime/blob/master/opencl/source/cl_device/cl_device_caps.cpp#L373-L389), so when they are not supported, the values returned by these members are undefined behavior / random numbers instead of clearly indicating lack of dp4a support by returning zero.

Please fix!

KamilKoprykIntel, 2025-03-16:
Hi @ProjectPhysX 
I don't think there is any UB there.
There is a value initialization for deviceInfo:
ClDeviceInfo deviceInfo = {};
which for fundamental types is a zero-initialization.

Best regards

ProjectPhysX, 2025-03-16:
Hi @KamilKoprykIntel,

I'm not sure about whether the code locations I've linked are the root cause of the UB, but I'm definitely observing UB on Windows 10 on UHD Graphics 630 and CPU:
```
Intel(R) Arc(TM) B580 Graphics: driver=32.0.101.6559, signed_accelerated=1, unsigned_accelerated=1, mixed_signedness_accelerated=1, accumulating_saturating_signed_accelerated=1, accumulating_saturating_unsigned_accelerated=1, accumulating_saturating_mixed_signedness_accelerated=1
Intel(R) UHD Graphics 630: driver=31.0.101.2130, signed_accelerated=509, unsigned_accelerated=509, mixed_signedness_accelerated=651817051, accumulating_saturating_signed_accelerated=32760, accumulating_saturating_unsigned_accelerated=0, accumulating_saturating_mixed_signedness_accelerated=0
Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz: driver=2024.18.10.0.08_160000, signed_accelerated=509, unsigned_accelerated=509, mixed_signedness_accelerated=651817051, accumulating_saturating_signed_accelerated=32760, accumulating_saturating_unsigned_accelerated=0, accumulating_saturating_mixed_signedness_accelerated=0
```
I'm using this code snippet for the printout:
```c
const auto idpap = cl_device.getInfo<CL_DEVICE_INTEGER_DOT_PRODUCT_ACCELERATION_PROPERTIES_4x8BIT_PACKED_KHR>();
println(name+": driver="+driver_version
	+", signed_accelerated="+to_string(idpap.unsigned_accelerated)
	+", unsigned_accelerated="+to_string(idpap.unsigned_accelerated)
	+", mixed_signedness_accelerated="+to_string(idpap.mixed_signedness_accelerated)
	+", accumulating_saturating_signed_accelerated="+to_string(idpap.accumulating_saturating_signed_accelerated)
	+", accumulating_saturating_unsigned_accelerated="+to_string(idpap.accumulating_saturating_unsigned_accelerated)
	+", accumulating_saturating_mixed_signedness_accelerated="+to_string(idpap.accumulating_saturating_mixed_signedness_accelerated)
);
```
Placing the above code snipped at a different location in source code returns different random numbers:
```
Intel(R) Arc(TM) B580 Graphics: driver=32.0.101.6559, signed_accelerated=1, unsigned_accelerated=1, mixed_signedness_accelerated=1, accumulating_saturating_signed_accelerated=1, accumulating_saturating_unsigned_accelerated=1, accumulating_saturating_mixed_signedness_accelerated=1
Intel(R) UHD Graphics 630: driver=31.0.101.2130, signed_accelerated=0, unsigned_accelerated=0, mixed_signedness_accelerated=766797825, accumulating_saturating_signed_accelerated=450, accumulating_saturating_unsigned_accelerated=3058693656, accumulating_saturating_mixed_signedness_accelerated=183
Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz: driver=2024.18.10.0.08_160000, signed_accelerated=0, unsigned_accelerated=0, mixed_signedness_accelerated=766797825, accumulating_saturating_signed_accelerated=450, accumulating_saturating_unsigned_accelerated=1, accumulating_saturating_mixed_signedness_accelerated=0
```

On my other Linux system it works without UB (except on Nvidia GPU):
```
Intel(R) Graphics [0xe20b]: driver=24.45.31740.9, signed_accelerated=1, unsigned_accelerated=1, mixed_signedness_accelerated=1, accumulating_saturating_signed_accelerated=1, accumulating_saturating_unsigned_accelerated=1, accumulating_saturating_mixed_signedness_accelerated=1
Intel(R) Arc(TM) A770 Graphics: driver=24.45.31740.9, signed_accelerated=1, unsigned_accelerated=1, mixed_signedness_accelerated=1, accumulating_saturating_signed_accelerated=1, accumulating_saturating_unsigned_accelerated=1, accumulating_saturating_mixed_signedness_accelerated=1
Intel(R) UHD Graphics 770: driver=24.45.31740.9, signed_accelerated=1, unsigned_accelerated=1, mixed_signedness_accelerated=1, accumulating_saturating_signed_accelerated=1, accumulating_saturating_unsigned_accelerated=1, accumulating_saturating_mixed_signedness_accelerated=1
13th Gen Intel(R) Core(TM) i7-13700K: driver=2024.18.10.0.08_160000, signed_accelerated=0, unsigned_accelerated=0, mixed_signedness_accelerated=0, accumulating_saturating_signed_accelerated=0, accumulating_saturating_unsigned_accelerated=0, accumulating_saturating_mixed_signedness_accelerated=0
NVIDIA TITAN Xp: driver=560.35.03, signed_accelerated=23670, unsigned_accelerated=23670, mixed_signedness_accelerated=0, accumulating_saturating_signed_accelerated=0, accumulating_saturating_unsigned_accelerated=1328461952, accumulating_saturating_mixed_signedness_accelerated=23670
```

Regards,
Moritz

HoppeMateusz, 2025-03-17:
Hi, it is very unlikely Intel compute runtime impacts caps reported by Nvidia driver.
Can you disable Intel graphics cards  and check what is reported for Nvidia?

ProjectPhysX, 2025-03-17:
The false reporting for Nvidia GPU is another bug in Nvidia driver, I've reported it to them.

Focus here is the UB on
- UHD Graphics 630 (Windows)
- i7-8700K (Windows)
- i7-13700K (Linux)

HoppeMateusz, 2025-03-17:
Are you checking the version of  cl_khr_integer_dot_product  extension  before printing ?
and the result of query to: cl_device.getInfo<CL_DEVICE_INTEGER_DOT_PRODUCT_ACCELERATION_PROPERTIES_4x8BIT_PACKED_KHR>()  ?

CL_DEVICE_INTEGER_DOT_PRODUCT_ACCELERATION_PROPERTIES_4x8BIT_PACKED_KHR  is only available starting from 2.0:

NOTE: [CL_DEVICE_INTEGER_DOT_PRODUCT_ACCELERATION_PROPERTIES_4x8BIT_PACKED_KHR](https://registry.khronos.org/OpenCL/specs/3.0-unified/html/OpenCL_API.html#CL_DEVICE_INTEGER_DOT_PRODUCT_ACCELERATION_PROPERTIES_4x8BIT_PACKED_KHR) is missing before version 2.0 of the extension.

Driver for :  Intel(R) UHD Graphics 630: driver=31.0.101.2134    may not have support for this new extension.

from :

```
Intel(R) Graphics [0xe20b]: driver=24.45.31740.9, CL_DEVICE_INTEGER_DOT_PRODUCT_CAPABILITIES_KHR=3, signed_accelerated=1, unsigned_accelerated=1, mixed_signedness_accelerated=1, accumulating_saturating_signed_accelerated=1, accumulating_saturating_unsigned_accelerated=1, accumulating_saturating_mixed_signedness_accelerated=1
Intel(R) Arc(TM) A770 Graphics: driver=24.45.31740.9, CL_DEVICE_INTEGER_DOT_PRODUCT_CAPABILITIES_KHR=3, signed_accelerated=1, unsigned_accelerated=1, mixed_signedness_accelerated=1, accumulating_saturating_signed_accelerated=1, accumulating_saturating_unsigned_accelerated=1, accumulating_saturating_mixed_signedness_accelerated=1
Intel(R) UHD Graphics 770: driver=24.45.31740.9, CL_DEVICE_INTEGER_DOT_PRODUCT_CAPABILITIES_KHR=3, signed_accelerated=1, unsigned_accelerated=1, mixed_signedness_accelerated=1, accumulating_saturating_signed_accelerated=1, accumulating_saturating_unsigned_accelerated=1, accumulating_saturating_mixed_signedness_accelerated=1
13th Gen Intel(R) Core(TM) i7-13700K: driver=2024.18.10.0.08_160000, CL_DEVICE_INTEGER_DOT_PRODUCT_CAPABILITIES_KHR=103331440451360, signed_accelerated=0, unsigned_accelerated=0, mixed_signedness_accelerated=0, accumulating_saturating_signed_accelerated=0, accumulating_saturating_unsigned_accelerated=0, accumulating_saturating_mixed_signedness_accelerated=0
NVIDIA TITAN Xp: driver=560.35.03, CL_DEVICE_INTEGER_DOT_PRODUCT_CAPABILITIES_KHR=103331438458800, signed_accelerated=0, unsigned_accelerated=0, mixed_signedness_accelerated=3124243104, accumulating_saturating_signed_accelerated=24058, accumulating_saturating_unsigned_accelerated=4, accumulating_saturating_mixed_signedness_accelerated=0
```

i see that newer Intel cards and drivers report correct values.

HoppeMateusz, 2025-03-17:
OpenCL driver for UHD 630 device  should return error from caps query ( clGetDeviceInfo() ) - old driver does not know new extension and new enum:   CL_DEVICE_INTEGER_DOT_PRODUCT_ACCELERATION_PROPERTIES_4x8BIT_PACKED_KHR  .  I suspect the structured is not written at all when error is returned - so random data comes from the caller.

Are errors handled in the code you are using ?

ProjectPhysX, 2025-03-17:
`clGetDeviceInfo` returns error -30 on UHD 630 👍
So only when clGetDeviceInfo doesn't error I can expect valid output. Closing this as application-side issue.
Thanks for the help!

closed 2025-03-17 by ProjectPhysX