intel/compute-runtime #817: __opencl_c_integer_dot_product_input_4x8bit / __opencl_c_integer_dot_product_input_4x8bit_packed feature macros are always enabled, even on unsupported devices

ProjectPhysX, 2025-03-12:
[`isDotIntegerProductExtensionSupported()` always returns true](https://github.com/intel/compute-runtime/blob/master/shared/source/helpers/compiler_product_helper_tgllp_and_later.inl#L12-L14), which is incorrect. It leads to the `__opencl_c_integer_dot_product_input_4x8bit` / `__opencl_c_integer_dot_product_input_4x8bit_packed` feature macros always being present, even on unsupported devices.

This means: If any OpenCL application implements dp4a with checks for `__opencl_c_integer_dot_product_input_4x8bit` / `__opencl_c_integer_dot_product_input_4x8bit_packed` feature macros to see of `dot()`/`dot_acc_sat()` functions are available, the application will instantly be broken on all older unsupported Intel platforms - compiling will just fail because `dot()`/`dot_acc_sat()` functions will be exposed by false feature macros but are not actually supported by compiler.

Please fix!

JablonskiMateusz, 2025-04-16:
@ProjectPhysX  
> This is especially important to fix on legacy driver too. 

For legacy platforms (pre-gen12) the function returns false: https://github.com/intel/compute-runtime/blob/releases/24.35/shared/source/helpers/compiler_product_helper_bdw_to_icllp.inl#L14

ProjectPhysX, 2025-04-26:
Hi @JablonskiMateusz,

I am observing that on Intel UHD 630 (i7-8700K) with driver 31.0.101.2134 (Windows), the `__opencl_c_integer_dot_product_input_4x8bit` and `__opencl_c_integer_dot_product_input_4x8bit_packed` feature macros are both present. How is this possible?

Regards,
Moritz

ProjectPhysX, 2025-10-01:
@pwilma https://opencl.gpuinfo.org/displayreport.php?id=4591