intel/compute-runtime #822: Can bundled CL headers be updated (v2024.10.24)?

frantisekz, 2025-03-30:
I am maintaining compute-runtime packages in Fedora and EPEL, and the packaging guidelines strongly prefer using system libraries and headers instead of the bundled ones.

I am building compute runtime with this flag to achieve use of system headers:
`-DKHRONOS_HEADERS_DIR="/usr/include/CL/"`

Use of the latest opencl-headers tag from upstream (v2024.10.24) requires a kind of hacky workaround to make the build and runtime use error-free: https://src.fedoraproject.org/rpms/intel-compute-runtime/blob/rawhide/f/0001-CL-Headers-2024.10.24.patch 

I understand that updating the bundled headers isn't a priority, but hope this reminder is welcome!

Thanks!

JablonskiMateusz, 2025-04-17:
updated headers in https://github.com/intel/compute-runtime/commit/21f6bb8889d293dbbfd808c41de56c49304765a3

frantisekz, 2025-04-26:
Thanks a lot!!

closed 2025-04-26 by frantisekz