intel/compute-runtime #826: Malformed L0 driver version string from github release packages

ph0b, 2025-04-11:
This issue got discovered and root caused initially from Blender: https://projects.blender.org/blender/blender/issues/137277

When using .deb packages from github releases, such as https://github.com/intel/compute-runtime/releases/tag/25.09.32961.5, zeIntelGetDriverVersionString which is supposed to report the driver version in the form: `Major.Minor.Patch+Optional` as defined in https://github.com/intel/compute-runtime/blob/master/level_zero/doc/experimental_extensions/GET_DRIVER_VERSION_STRING.md, doesn't. 

Instead of `1.6.32961+5` for this recent release, it returns `1.6.32961.500000`.
This issue is present since at least https://github.com/intel/compute-runtime/releases/tag/24.39.31294.12

Since `std::to_string(32961.5)` is `32961.500000`, my analysis is that instead of being built with `NEO_VERSION_BUILD=32961` and `NEO_VERSION_HOTFIX=5`, the packages are built with `NEO_VERSION_BUILD=32961.5` and `NEO_VERSION_HOTFIX=0`.

nsirgien, 2025-06-18:
Fixed in https://github.com/intel/compute-runtime/commit/fba685008bf528b56dee5319782cdb11eb1ac75c, as far as I know.

JablonskiMateusz, 2025-08-21:
@ph0b  could you confirm the latest release works fine?>

JablonskiMateusz, 2025-09-02:
@ph0b  thanks for catching that

JablonskiMateusz, 2025-09-04:
@ph0b  change is re-applied to release 25.35 branch, should be available in the next release https://github.com/intel/compute-runtime/commit/68d43aafe612c7409d9e9984a658d88d527ce8dc

ph0b, 2025-09-16:
I can confirm the issue is fixed with today's release [25.35.35096.9](https://github.com/intel/compute-runtime/releases/tag/25.35.35096.9), thank you!
```
[level_zero:gpu][level_zero:0] Intel(R) oneAPI Unified Runtime over Level-Zero, Intel(R) Graphics [0xe20b] 20.1.0 [1.6.35096+9]
[level_zero:gpu][level_zero:1] Intel(R) oneAPI Unified Runtime over Level-Zero, Intel(R) Graphics [0xa780] 12.2.0 [1.6.35096+9]
```