intel/compute-runtime #828: runtime currently relies on deprecated i915 linux driver and uapi support

ct-clmsn, 2025-05-07:
Both `level_zero` and the `shared/source/os_instance/linux` only use the deprecated `i915` driver and uapi support instead of the `xe` driver and uapi support. The runtime has some support for `xe` but that support has to been used in the implementation.

`shared/source/os_interface/linux/ioctl_helper.h` implements `IoctlHelperI915`, `IoctlHelperUpstream`, `IoctlHelperImpl`, `IoctlHelperPrelim20` all use the `i915` support but there does not exist an `IoctlHelperXe` that uses `xe` support.

JablonskiMateusz, 2025-05-07:
Hi @ct-clmsn 
`IoctlHelperXe` implements `IoctlHelper` interface and it is available in default driver build. In case of `xe` linux driver (like for Lunarlake/Battlemage) we don't call any i915-specific ioctl

ct-clmsn, 2025-05-07:
@JablonskiMateusz ah wait, I `IoctlHelperXe ` and specifically the `Drm::setupIoctlHelper` has a conditional that checks for `xe`. My apologies!

closed 2025-05-07 by ct-clmsn