intel/compute-runtime #829: Unable to install on Ubuntu 24.04 after package version change

martygrant, 2025-05-09:
I tried to install the Compute Runtime on a fresh Ubuntu 24.04 install following the instructions on the [releases](https://github.com/intel/compute-runtime/releases/) page but was unable to successfully install the OpenCL component.

The instruction process would show no obvious error but running `clinfo` or `sycl-ls` or `urinfo` from DPC++ would not list the Intel OpenCL driver. No ICD file was added to `/etc/OpenCL/vendors`.

It seems like something has changed since the package version has changed i.e. `intel-igc-core` to `intel-igc-core-2`. 

Is this a known issue? Can you recommend a workaround or another installation method?

fivemeyestore, 2025-05-09:
Great implementation of this feature. I particularly like the approach to error handling.

JablonskiMateusz, 2025-05-12:
Hi @martygrant  
could you please share installation logs?

pwilma, 2025-09-25:
Hi @martygrant 

We’d like to know if this issue is still affecting you. If so, please provide an update or any additional information. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated!