intel/compute-runtime #832: No [level_zero:gpu] (only [opencl:gpu]) for i7-11700K iGPU

owlshrimp, 2025-06-05:
Hello,

I have an i7-11700K system with a rocketlake processor. I'm running Ubuntu 24.04.2 LTS x86_64 with kernel 6.8.0-60-generic. I followed the following steps to install the oneAPI base toolkit:
```
$ sudo apt-get install intel-opencl-icd
$ sudo clinfo -l
Platform #0: NVIDIA CUDA
 `-- Device #0: NVIDIA GeForce RTX 3090
Platform #1: Intel(R) OpenCL Graphics
 `-- Device #0: Intel(R) UHD Graphics 750
$ sudo usermod -aG render username
$ sudo usermod -aG video username

$ sudo apt update
$ sudo apt install -y gpg-agent wget # -> already installed, not needed
$ wget -O- https://apt.repos.intel.com/intel-gpg-keys/GPG-PUB-KEY-INTEL-SW-PRODUCTS.PUB | gpg --dearmor | sudo tee /usr/share/keyrings/oneapi-archive-keyring.gpg > /dev/null
$ echo "deb [signed-by=/usr/share/keyrings/oneapi-archive-keyring.gpg] https://apt.repos.intel.com/oneapi all main" | sudo tee /etc/apt/sources.list.d/oneAPI.list

$ sudo apt update
$ sudo apt install intel-oneapi-base-toolkit
# -> compiled a bunch of drivers (socperf sep pax vtsspp socwatch) as well as a sep5.service
```
However, running:
```
$ source /opt/intel/oneapi/setvars.sh
$ sycl-ls
```
shows:
```
[opencl:cpu][opencl:0] Intel(R) OpenCL, 11th Gen Intel(R) Core(TM) i7-11700K @ 3.60GHz OpenCL 3.0 (Build 0) [2025.19.4.0.18_160000.xmain-hotfix]
[opencl:gpu][opencl:1] Intel(R) OpenCL Graphics, Intel(R) UHD Graphics 750 OpenCL 3.0 NEO  [23.43.027642]```
Where there is no [level_zero:gpu] entry for the iGPU as [1] suggests should be there:
```
When targeting an intel GPU, the user should expect one or more devices among the available SYCL devices. Please make sure that at least one GPU is present via sycl-ls, for instance [level_zero:gpu] in the sample output below:
[opencl:acc][opencl:0] Intel(R) FPGA Emulation Platform for OpenCL(TM), Intel(R) FPGA Emulation Device OpenCL 1.2  [2023.16.10.0.17_160000]
[opencl:cpu][opencl:1] Intel(R) OpenCL, 13th Gen Intel(R) Core(TM) i7-13700K OpenCL 3.0 (Build 0) [2023.16.10.0.17_160000]
[opencl:gpu][opencl:2] Intel(R) OpenCL Graphics, Intel(R) Arc(TM) A770 Graphics OpenCL 3.0 NEO  [23.30.26918.50]
[level_zero:gpu][level_zero:0] Intel(R) Level-Zero, Intel(R) Arc(TM) A770 Graphics 1.3 [1.3.26918]
```
Also attached is a log from installing the intel-oneapi-base-toolkit package [2] and the output of `SYCL_UR_TRACE=2 strace -o output.log sycl-ls` at [3].

Are there additional steps required to get the [level_zero:gpu] device to show up? My understanding is the 11th gen rocketlake iGPUs are currently supported.

[1] https://github.com/ggml-org/llama.cpp/blob/0d3984424f2973c49c4bcabe4cc0153b4f90c601/docs/backend/SYCL.md
[2] https://pastebin.com/fay5X3cW
[3] https://pastebin.com/L2akg2Tv

owlshrimp, 2025-06-06:
```
libze1/noble 1.16.1-1build1 amd64
  oneAPI Level Zero -- share libraries
```
```
$ sudo apt-get install libze1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  libze1
0 upgraded, 1 newly installed, 0 to remove and 25 not upgraded.
Need to get 212 kB of archives.
After this operation, 1,320 kB of additional disk space will be used.
Get:1 http://ca.archive.ubuntu.com/ubuntu noble/universe amd64 libze1 amd64 1.16.1-1build1 [212 kB]
Fetched 212 kB in 0s (626 kB/s)
Selecting previously unselected package libze1:amd64.
(Reading database ... 359561 files and directories currently installed.)
Preparing to unpack .../libze1_1.16.1-1build1_amd64.deb ...
Unpacking libze1:amd64 (1.16.1-1build1) ...
Setting up libze1:amd64 (1.16.1-1build1) ...
Processing triggers for libc-bin (2.39-0ubuntu8.4) ...
```
doesn't seem to have done much of anything

owlshrimp, 2025-06-06:
```$ ZE_ENABLE_LOADER_DEBUG_TRACE=1 sycl-ls
ZE_LOADER_DEBUG_TRACE:Loading Driver libze_intel_gpu.so.1
ZE_LOADER_DEBUG_TRACE:Load Library of libze_intel_gpu.so.1 failed with libze_intel_gpu.so.1: cannot open shared object file: No such file or directory
ZE_LOADER_DEBUG_TRACE:Loading Driver libze_intel_vpu.so.1
ZE_LOADER_DEBUG_TRACE:Load Library of libze_intel_vpu.so.1 failed with libze_intel_vpu.so.1: cannot open shared object file: No such file or directory
[opencl:cpu][opencl:0] Intel(R) OpenCL, 11th Gen Intel(R) Core(TM) i7-11700K @ 3.60GHz OpenCL 3.0 (Build 0) [2025.19.4.0.18_160000.xmain-hotfix]
[opencl:gpu][opencl:1] Intel(R) OpenCL Graphics, Intel(R) UHD Graphics 750 OpenCL 3.0 NEO  [23.43.027642]
```
Maybe it's a driver issue?

owlshrimp, 2025-06-06:
running with different args showed:
```
$ ZE_ENABLE_LOADER_DEBUG_TRACE=1 sycl-ls
ZE_LOADER_DEBUG_TRACE:Loading Driver libze_intel_gpu.so.1
ZE_LOADER_DEBUG_TRACE:Load Library of libze_intel_gpu.so.1 failed with libze_intel_gpu.so.1: cannot open shared object file: No such file or directory
ZE_LOADER_DEBUG_TRACE:Loading Driver libze_intel_vpu.so.1
ZE_LOADER_DEBUG_TRACE:Load Library of libze_intel_vpu.so.1 failed with libze_intel_vpu.so.1: cannot open shared object file: No such file or directory
[opencl:cpu][opencl:0] Intel(R) OpenCL, 11th Gen Intel(R) Core(TM) i7-11700K @ 3.60GHz OpenCL 3.0 (Build 0) [2025.19.4.0.18_160000.xmain-hotfix]
[opencl:gpu][opencl:1] Intel(R) OpenCL Graphics, Intel(R) UHD Graphics 750 OpenCL 3.0 NEO  [23.43.027642]
```
so `libze_intel_gpu.so.1` was missing.

Searching with `apt search libze` turned up:
```
libze-dev/noble 1.16.1-1build1 amd64
  oneAPI Level Zero -- development files

libze-intel-gpu-dev/noble 23.43.27642.40-1ubuntu3 amd64
  Intel oneAPI L0 support implementation for Intel GPUs -- shared library

libze-intel-gpu1/noble 23.43.27642.40-1ubuntu3 amd64
  Intel oneAPI L0 support implementation for Intel GPUs -- shared library

libze1/noble,now 1.16.1-1build1 amd64 [installed]
  oneAPI Level Zero -- share libraries
```
and
```
$ sudo apt-get install libze-intel-gpu1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  libze-intel-gpu1
0 upgraded, 1 newly installed, 0 to remove and 25 not upgraded.
Need to get 2,332 kB of archives.
After this operation, 28.8 MB of additional disk space will be used.
Get:1 http://ca.archive.ubuntu.com/ubuntu noble/universe amd64 libze-intel-gpu1 amd64 23.43.27642.40-1ubuntu3 [2,332 kB]
Fetched 2,332 kB in 0s (5,261 kB/s)        
Selecting previously unselected package libze-intel-gpu1.
(Reading database ... 359561 files and directories currently installed.)
Preparing to unpack .../libze-intel-gpu1_23.43.27642.40-1ubuntu3_amd64.deb ...
Unpacking libze-intel-gpu1 (23.43.27642.40-1ubuntu3) ...
Setting up libze-intel-gpu1 (23.43.27642.40-1ubuntu3) ...
Processing triggers for libc-bin (2.39-0ubuntu8.4) ...
```
Seems to have fixed things:
```
$ ZE_ENABLE_LOADER_DEBUG_TRACE=1 sycl-ls
ZE_LOADER_DEBUG_TRACE:Loading Driver libze_intel_gpu.so.1
ZE_LOADER_DEBUG_TRACE:Loading Driver libze_intel_vpu.so.1
ZE_LOADER_DEBUG_TRACE:Load Library of libze_intel_vpu.so.1 failed with libze_intel_vpu.so.1: cannot open shared object file: No such file or directory
ZE_LOADER_DEBUG_TRACE:check_drivers(flags=ZE_INIT_FLAG_GPU_ONLY)
ZE_LOADER_DEBUG_TRACE:init driver libze_intel_gpu.so.1 zeInit(ZE_INIT_FLAG_GPU_ONLY) returning ZE_RESULT_SUCCESS
[level_zero:gpu][level_zero:0] Intel(R) oneAPI Unified Runtime over Level-Zero, Intel(R) UHD Graphics 750 12.1.0 [1.3.27642]
[opencl:cpu][opencl:0] Intel(R) OpenCL, 11th Gen Intel(R) Core(TM) i7-11700K @ 3.60GHz OpenCL 3.0 (Build 0) [2025.19.4.0.18_160000.xmain-hotfix]
[opencl:gpu][opencl:1] Intel(R) OpenCL Graphics, Intel(R) UHD Graphics 750 OpenCL 3.0 NEO  [23.43.027642]
```
at least insofar that there's now a [level_zero:gpu] entry.

owlshrimp, 2025-06-06:
Both `libze1` and `libze-intel-gpu1` ended up being needed for that.

owlshrimp, 2025-06-12:
> > Both `libze1` and `libze-intel-gpu1` ended up being needed for that.
> 
> That's right. `intel-opencl-icd` is just for OCL, for level zero you need `libze1` (L0 loader) and `libze-intel-gpu1` (L0 GPU runtime)

I'm surprised they aren't dependencies of the official APT package I installed.

owlshrimp, 2025-06-30:
Redacted some personal information in logs.

eero-t, 2025-06-30:
>>> Both `libze1` and `libze-intel-gpu1` ended up being needed for that.
>>
>> That's right. `intel-opencl-icd` is just for OCL, for level zero you need `libze1` (L0 loader) and `libze-intel-gpu1` (L0 GPU runtime)
> 
> I'm surprised they aren't dependencies of the official APT package I installed.

You mean `intel-oneapi-base-toolkit`?  That's a good point.  'm not sure where the issues related to those higher level libraries & packages should be reported.  @JablonskiMateusz any ideas?

owlshrimp, 2025-09-25:
> [@owlshrimp](https://github.com/owlshrimp) could you point which instructions were you following?
> 
> Based on the link you shared, it looks like you missed the `Install the Driver Packages` section
> 
> [image]

There seem to be 4 empty sections at https://www.intel.com/content/www/us/en/developer/tools/oneapi/base-toolkit-download.html?packages=oneapi-toolkit&oneapi-toolkit-os=linux&oneapi-lin=apt including one that might have had instructions for installing additional driver packages?

kgibala, 2025-11-28:
Hi @owlshrimp,

This looks like it might be a browser incompatibility issue. I checked on Opera Browser and the content is properly displayed there.

[image]

eero-t, 2025-11-28:
> I see the same problem in both ubuntu's official chromium snap and firefox, so this may be a fairly severe compatibility issue.

Works both in Firefox (v145.0), and Chromium (v142.0) in here, but you need to allow cookies.  Site seems to be using cookies to keep track which of the menus is open.