intel/compute-runtime #833: Clarification for Ubuntu 24.10 over 24.04/25.04

arguellocarlos, 2025-06-28:
According to the compute runtime release notes for Battlemage users: _For the Lunar Lake and Battlemage platforms, the best experience is achieved using Ubuntu 24.10 with the intel-graphics PPA. See_ [https://dgpu-docs.intel.com/driver/client/overview.html#ubuntu-latest](https://dgpu-docs.intel.com/driver/client/overview.html#ubuntu-latest).

However, since support for Ubuntu 24.10 ends in July and the intel-graphics PPA is also available for Ubuntu 25.04, would it be better to upgrade to 25.04 or revert to the LTS version, 24.04?"

On that note, are there any advantages to using the ```linux-intel``` kernel from the intel-graphics PPA compared to the default Ubuntu stock kernel?

Thanks.

PiotrRozenfeld, 2025-07-24:
You are correct, Ubuntu 24.10 support ends in July. We have removed the comment about recommending 24.10 from the most recent release published last week. Both upgrade options you outlined are available – for our internal validation we have decided to use 24.04.2 LTS for consistency with other platforms, for typical end-users using Ubuntu 25.04 might be a slightly better option.

linux-intel kernel from the intel-graphics PPA offers moderate improvements over Ubuntu stock kernel, mostly bugfixes and EU debugging capability. For typical usage, either is adequate. Note that stock kernel in Ubuntu 24.04 does not contain LNL & BMG support (yet).

kgibala, 2026-01-16:
Hi @arguellocarlos,

We're closing this issue due to inactivity. We haven't received the requested information or updates for an extended period.

If you encounter this or a similar problem in the future, please feel free to create a new issue. When submitting, please follow our [Issue Submission Guide](https://github.com/intel/compute-runtime/blob/master/ISSUE_SUBMISSION_GUIDE.md) to help us understand and address your concern efficiently.