intel/compute-runtime #839: NEO::DirectSubmissionHw memset/memcpy error

ct-clmsn, 2025-07-29:
Related to #821 ; Demonstration programs are generating a memcpy error at runtime. It's unclear what is going on - advice on debugging this issue would be appreciated.

```
#include <sycl/sycl.hpp>

int main(int argc, char* argv[]) {
  sycl::queue q;

  std::cout << "Running on "
            << q.get_device().get_info<sycl::info::device::name>()
            << "\n";
}
```

```
ONEAPI_DEVICE_SELECTOR=opencl:gpu ./hellosycl0
```

```
Running on Intel(R) Arc(TM) A750 Graphics

Thread 1 "hellosycl0" received signal SIGBUS, Bus error.
Download failed: Invalid argument.  Continuing without source file ./string/./string/memset.c.
__GI_memset (dstpp=0x3ff3baf000, c=0, len=<optimized out>) at ./string/memset.c:56
warning: 56 ./string/memset.c: No such file or directory
(gdb) bt
#0  __GI_memset (dstpp=0x3ff3baf000, c=0, len=<optimized out>) at ./string/memset.c:56
#1  0x0000003ff488e446 in NEO::DirectSubmissionHw<NEO::XeHpgCoreFamily, NEO::RenderDispatcher<NEO::XeHpgCoreFamily> >::allocateResources() ()
   from libigdrcl.so
#2  0x0000003ff488e22e in NEO::DirectSubmissionHw<NEO::XeHpgCoreFamily, NEO::RenderDispatcher<NEO::XeHpgCoreFamily> >::initialize(bool) ()
   from libigdrcl.so
#3  0x0000003ff47ee4ae in NEO::CommandStreamReceiverHw<NEO::XeHpgCoreFamily>::initDirectSubmission() () from libigdrcl.so
#4  0x0000003ff4632984 in NEO::CommandQueue::initializeGpgpuInternals() const () from libigdrcl.so
#5  0x0000003ff4632812 in NEO::CommandQueue::initializeGpgpu() const () from libigdrcl.so
#6  0x0000003ff46329f6 in NEO::CommandQueue::getGpgpuCommandStreamReceiver() const () from libigdrcl.so
#7  0x0000003ff46d29b0 in NEO::CommandQueueHw<NEO::XeHpgCoreFamily>::flush() () from libigdrcl.so
#8  0x0000003ff45d07ec in clReleaseCommandQueue () from libigdrcl.so
#9  0x0000003ff65c5054 in clReleaseCommandQueue () from libOpenCL.so.1
#10 0x0000003ff6f3d576 in urQueueRelease () from libur_adapter_opencl.so.0
#11 0x0000003ff70672b8 in urQueueRelease () from libur_loader.so.0
#12 0x0000003ff727d32e in sycl::_V1::detail::queue_impl::~queue_impl() () from libsycl.so.8
#13 0x0000000000013486 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)1>::_M_release_last_use (this=0x12c34a0)
    at /usr/lib/gcc/riscv64-linux-gnu/13/../../../../include/c++/13/bits/shared_ptr_base.h:175
#14 0x0000000000013462 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)1>::_M_release (this=0x12c34a0)
    at /usr/lib/gcc/riscv64-linux-gnu/13/../../../../include/c++/13/bits/shared_ptr_base.h:311
#15 0x00000000000133aa in std::__shared_count<(__gnu_cxx::_Lock_policy)1>::~__shared_count (this=0x3fffffee68)
    at /usr/lib/gcc/riscv64-linux-gnu/13/../../../../include/c++/13/bits/shared_ptr_base.h:1071
#16 0x0000000000013806 in std::__shared_ptr<sycl::_V1::detail::queue_impl, (__gnu_cxx::_Lock_policy)1>::~__shared_ptr (this=0x3fffffee60)
    at /usr/lib/gcc/riscv64-linux-gnu/13/../../../../include/c++/13/bits/shared_ptr_base.h:1524
#17 0x00000000000137e8 in std::shared_ptr<sycl::_V1::detail::queue_impl>::~shared_ptr (this=0x3fffffee60)
    at /usr/lib/gcc/riscv64-linux-gnu/13/../../../../include/c++/13/bits/shared_ptr.h:175
#18 0x000000000001240a in sycl::_V1::queue::~queue (this=0x3fffffee60) at llvm-6.1.0/build/bin/../include/sycl/queue.hpp:132
#19 0x00000000000121d8 in main (argc=1, argv=0x3ffffff028) at hellosycl.cpp:9
(gdb) q
```

```
#include <sycl/sycl.hpp>

#define N 100

int main(int argc, char* argv[]) {
  sycl::queue q;

  std::cout << "Running on "
            << q.get_device().get_info<sycl::info::device::name>()
            << "\n";

  int aData[N];
  int bData[N];
[... 46 lines cut ...]
#14 0x0000003ff73f4c7a in sycl::_V1::detail::ProgramManager::getBuiltURProgram(sycl::_V1::detail::ObjectWithDeps<sycl::_V1::detail::RTDeviceBinaryImage const*> const&, sycl::_V1::context const&, std::vector<sycl::_V1::device, std::allocator<sycl::_V1::device> > const&, sycl::_V1::detail::ObjectWithDeps<sycl::_V1::detail::device_image_plain> const*, std::vector<unsigned char, std::allocator<unsigned char> > const&) () from llvm-6.1.0/build/lib/libsycl.so.8
#15 0x0000003ff73f6632 in sycl::_V1::detail::ProgramManager::getBuiltURProgram(std::shared_ptr<sycl::_V1::detail::context_impl> const&, std::shared_ptr<sycl::_V1::detail::device_impl> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, sycl::_V1::detail::NDRDescT const&, bool) ()
   from llvm-6.1.0/build/lib/libsycl.so.8
#16 0x0000003ff73fce44 in sycl::_V1::detail::ProgramManager::getOrCreateKernel(std::shared_ptr<sycl::_V1::detail::context_impl> const&, std::shared_ptr<sycl::_V1::detail::device_impl> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, sycl::_V1::detail::NDRDescT const&) ()
   from llvm-6.1.0/build/lib/libsycl.so.8
#17 0x0000003ff742ce62 in sycl::_V1::detail::enqueueImpKernel(std::shared_ptr<sycl::_V1::detail::queue_impl> const&, sycl::_V1::detail::NDRDescT&, std::vector<sycl::_V1::detail::ArgDesc, std::allocator<sycl::_V1::detail::ArgDesc> >&, std::shared_ptr<sycl::_V1::detail::kernel_bundle_impl> const&, std::shared_ptr<sycl::_V1::detail::kernel_impl> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<ur_event_handle_t_*, std::allocator<ur_event_handle_t_*> >&, std::shared_ptr<sycl::_V1::detail::event_impl> const&, std::function<void* (sycl::_V1::detail::AccessorImplHost*)> const&, ur_kernel_cache_config_t, bool, bool, unsigned long, sycl::_V1::detail::RTDeviceBinaryImage const*) () from libsycl.so.8
#18 0x0000003ff74325f4 in sycl::_V1::detail::ExecCGCommand::enqueueImpQueue() () from libsycl.so.8
#19 0x0000003ff7426b92 in sycl::_V1::detail::Command::enqueue(sycl::_V1::detail::EnqueueResultT&, sycl::_V1::detail::BlockingT, std::vector<sycl::_V1::detail::Command*, std::allocator<sycl::_V1::detail::Command*> >&) () from libsycl.so.8
#20 0x0000003ff743b6b4 in sycl::_V1::detail::Scheduler::GraphProcessor::enqueueCommand(sycl::_V1::detail::Command*, std::shared_lock<std::shared_timed_mutex>&, sycl::_V1::detail::EnqueueResultT&, std::vector<sycl::_V1::detail::Command*, std::allocator<sycl::_V1::detail::Command*> >&, sycl::_V1::detail::Command*, sycl::_V1::detail::BlockingT) [clone .part.0] ()
   from libsycl.so.8
#21 0x0000003ff7438fec in sycl::_V1::detail::Scheduler::enqueueCommandForCG(std::shared_ptr<sycl::_V1::detail::event_impl>, std::vector<sycl::_V1::detail::Command*, std::allocator<sycl::_V1::detail::Command*> >&, sycl::_V1::detail::BlockingT) () from llvm-6.1.0/build/lib/libsycl.so.8
#22 0x0000003ff743b00a in sycl::_V1::detail::Scheduler::addCG(std::unique_ptr<sycl::_V1::detail::CG, std::default_delete<sycl::_V1::detail::CG> >, std::shared_ptr<sycl::_V1::detail::queue_impl> const&, bool, ur_exp_command_buffer_handle_t_*, std::vector<unsigned int, std::allocator<unsigned int> > const&) ()
   from llvm-6.1.0/build/lib/libsycl.so.8
#23 0x0000003ff746c98a in sycl::_V1::handler::finalize() () from /home/ct/Downloads/llvm-6.1.0/build/lib/libsycl.so.8
#24 0x0000003ff740262c in void sycl::_V1::detail::queue_impl::finalizeHandler<sycl::_V1::handler>(sycl::_V1::handler&, sycl::_V1::event&) ()
   from llvm-6.1.0/build/lib/libsycl.so.8
#25 0x0000003ff7402e40 in sycl::_V1::detail::queue_impl::submit_impl(std::function<void (sycl::_V1::handler&)> const&, std::shared_ptr<sycl::_V1::detail::queue_impl> const&, std::sha--Type <RET> for more, q to quit, c to continue without paging--
red_ptr<sycl::_V1::detail::queue_impl> const&, std::shared_ptr<sycl::_V1::detail::queue_impl> const&, bool, sycl::_V1::detail::code_location const&, bool, sycl::_V1::detail::SubmissionInfo const&) () from libsycl.so.8
#26 0x0000003ff74871b8 in sycl::_V1::queue::submit_with_event_impl(std::function<void (sycl::_V1::handler&)>, sycl::_V1::detail::SubmissionInfo const&, sycl::_V1::detail::code_location const&, bool) () from libsycl.so.8
#27 0x0000000000013fb0 in sycl::_V1::queue::submit_with_event<main::{lambda(auto:1&)#1}>(main::{lambda(auto:1&)#1}, sycl::_V1::queue*, sycl::_V1::detail::code_location const&) (
    this=0x3fffffee58, CGF=..., SecondaryQueuePtr=0x0, CodeLoc=...) at /include/sycl/queue.hpp:2826
#28 0x0000000000013edc in sycl::_V1::queue::submit<main::{lambda(auto:1&)#1}>(main::{lambda(auto:1&)#1}, sycl::_V1::detail::code_location const&) (this=0x3fffffee58, CGF=...,
    CodeLoc=...) at llvm-6.1.0/build/bin/../include/sycl/queue.hpp:365
#29 0x0000000000013dda in main (argc=1, argv=0x3ffffff028) at hellosycl1.cpp:18
```

ct-clmsn, 2025-08-11:
Thank you for getting back quickly; I've been on travel recently and have been catching up!

kgibala, 2025-11-26:
Hi @ct-clmsn,

We would like to know if you are still intent to work on this PR https://github.com/intel/compute-runtime/pull/821.  If so, please provide an update or any additional information. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated.